PX4/PX4-Autopilot #9901: Stale mag data not detected

ndepal, 2018-07-13:
We just had a series of flights with marginally stable position control performance (toilet bowling). Analysis of the logs showed that the mag measurements (external HMC5983, internal mags disabled) are not changing one bit. This was only fixed by power cycling the vehicle, rebooting the FC did not change anything.

There are two problems with this incident:
1) The mag malfunctioned. Has anybody seen something like this before?
2) More importantly: Why was this not detected by the flight controller? Should this data not be marked stale / timed out and arming in position controlled modes/switching to these modes refused?

In addition to the mag measurements being constant, EKF2's heading innovations (configured to heading only fusion) are constant, i.e. no mag measurements are fused. This makes sense considering that the mag measurements are old (`sensor_combined_0.magnetometer_timestamp_relative + sensor_combined_0.timestamp` is constant). But the heading / position estimate is not marked invalid by EKF2.

Our firmware is based on 1.7.4.

Here is a log file: https://logs.px4.io/plot_app?log=8f1505e6-f998-47e4-9693-58c3c0424774

Any help would be appreciated!

dagar, 2018-07-13:
I think the core problem here is that the copy of sensor_combined inside sensors still has the old mag data and continues to publish. In more recent Firmware the magnetometer has been moved from the sensor_combined message to vehicle_magnetometer.

We should also review the specifics in the failover code for the case when there's only a single sensor. https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L898 
It should still be notifying you of stale data.

ndepal, 2018-07-13:
Right, `sensor_combined` has the same old mag data, but also with the same old timestamp. So the consumers of `sensor_combined` can tell that the data is old.

My question is, who is responsible for signaling a mag timeout and why did that not happen?

dagar, 2018-07-13:
> My question is, who is responsible for signaling a mag timeout and why did that not happen?
See my 2nd paragraph above.

I'm not sure how hard it is for you to sync up, but I would consider verifying this works correctly under v1.8.0+ where vehicle_magnetometer should simply stop publishing.

ndepal, 2018-07-13:
It's non-trivial to sync up, and it's also very difficult to reproduce this issue, we've never seen it happen before.

It's still it's not quite clear to me at which stage the failure happened (in the mag itself, I2C read, driver ...).

It looks like if `VotedSensorsUpdate` didn't get any new messages on the `sensor_mag` topic it would have signaled a timeout. It doesn't seem like that happened (certainly not in this log, but I also have no evidence of it happening in previous flights or in between flights). I guess since we're only logging `sensor_combined` there is no telling whether there was still data published on `sensor_mag`.

Since all internal mags are disabled, there is no mag to switch to. What would should happen if the only mag times out?

LorenzMeier, 2018-07-13:
Without any fallback option it would of course stick to the current mag. It is then up to the EKF to detect the stuck sensor.

ndepal, 2018-07-13:
Are you saying that this is an existing feature in 1.7.4 that didn't work or a needed feature?

ndepal, 2018-07-24:
As per @dagar 's suggestion, I have opened an issue to discuss the EKF behavior here: https://github.com/PX4/ecl/issues/479

@LorenzMeier and @priseborough provide conflicting suggestions about where the fault should be handled.

I see three problems:
1. The mag timeout was not detected at all in the log linked above. When trying to reproduce the problem as described here https://github.com/PX4/ecl/issues/479#issuecomment-405588613, I was not able to cause the mag timestamps to stop updating without the sensors module noticing and printing `Mag #0 fail:  TIMEOUT!`. So I may not be simulating the exact scenario form the log.
2. With the changes to reproduce the issue https://github.com/ndepal/Firmware/commit/734167dd3d27a3b42b7d21b269f805ede32e184c, `sensors` declares a `Mag #0 fail:  TIMEOUT!`. But it still allows arming/switching to position control. SITL only has 1 mag, right (i.e. it can't fail over to any other mag)? Is this the intended behavior? I would want the system to refuse arming/go into failsafe mode.
3. With the changes to flag position invalid in EKF https://github.com/ndepal/ecl/commit/a15bce935167b0ba3a8a4967bc12d6f124d1293a EKF2 does detect the timeout, sets `_fault_status.flags.bad_mag_hdg` to `true`, and `estimator_status.solution_status_flags` goes to 192 (i.e. neither `True if the horizontal position (relative) estimate is good` nor `True if the horizontal position (absolute) estimate is good`), but still no failsafe mode is entered. This may be due to my lack of understanding when failsafes are entered. Could somebody help me out here?

At the end of the day, it appears to me that it is possible to fly in position modes without a functioning mag. This is problematic for our application, if not for all multicopters.
The functionality I need is:
If the one and only mag is faulty (either timed out or provides measurements which are repeatedly rejected), detect it and either refuse arming in position modes or go to failsafe if already armed.

CarlOlsson, 2018-09-05:
> With the changes to reproduce the issue ndepal@734167d, sensors declares a Mag #0 fail: TIMEOUT!. But it still allows arming/switching to position control. SITL only has 1 mag, right (i.e. it can't fail over to any other mag)? Is this the intended behavior? I would want the system to refuse arming/go into failsafe mode.

This looks like another case of the [similar issue for the barometer](https://github.com/PX4/Firmware/issues/8465)

> The mag malfunctioned. Has anybody seen something like this before?

Yes we just hit something very similar, the magnetometer stopped giving data in the middle of the flight
[image]

> We should also review the specifics in the failover code for the case when there's only a single sensor. https://github.com/PX4/Firmware/blob/master/src/modules/sensors/voted_sensors_update.cpp#L898
It should still be notifying you of stale data.

Yes and the data should then not reach the EKF. In our case it was still trying to fuse the mag even though the measurements were constant
[image]

> If the one and only mag is faulty (either timed out or provides measurements which are repeatedly rejected), detect it and either refuse arming in position modes or go to failsafe if already armed.

Regarding the "or provides measurements which are repeatedly rejected". I think we should extend this to all sensors. Somewhere (e.g. in commander) the EKF performance should be monitored and if anomalies are detected a parameter should decide what to do:
E.g. nothing, report over mavlink, return to home, failsafe

The conditions could be many things but e.g:
- a sensor has been rejected for an accumulated time of more than TBD seconds
- the mag is declared faulty inside the EKF
- estimator_status.filter_fault_flags > 0
- estimator_status.vibe > TBD
- etc

Without this the system will keep on flying even when severe anomalies are detected by the EKF.

closed 2019-02-04 by stale[bot]