PX4/PX4-Autopilot #9904: Problems with irlock integration in px4

Andree512, 2018-07-13:
Hello!
I have a problem with pixy integration in px4, performed all the checks and settings according to the manual https://docs.px4.io/en/advanced_features/precland.html, when the drones are started with the pixy (irlock) + lidar module enabled, the QGroundControle parameters appear points "Landing target estimator" and "Precision land". However, the drone does not land on the ground beacon, tried different scenarios, sat down with the commander command mode auto: precland, and also in the mission.

ndepal, 2018-07-17:
Can you share a log?

Andree512, 2018-07-25:
@ndepal 

[log.zip](https://github.com/PX4/Firmware/files/2227958/log.zip)

dagar, 2018-07-25:
https://logs.px4.io/plot_app?log=837d0b9e-43d7-4d63-abd4-3376de577a82

ndepal, 2018-07-25:
@Andree512 Your logs are missing almost all fields. E.g. `irlock_report` and `vehicle_status`. Why is this the case?
I see `landing_target_pose` messages in the first log, but not the second. So it looks like the setup is correct. Because of the missing log data, I can't tell what is happening and supposed to happen as far as flight modes are concerned.

Andree512, 2018-07-26:
@ndepal 
I correctly understand that to solve this problem it is necessary to irlock_report and vehicle_status? Please tell me how to activate them.
And we were able to landing it exactly on the beacon, as follows: through the console in qgrouncontrlo by the command "commander mode auto: precland", after two seconds we translate it with RC in "Position mode", then again "commander mode auto: precland "and so every 2 seconds to the very ground.

ndepal, 2018-07-26:
These topics would help me understand what's going on. Have you configured custom topics to be logged as described here https://dev.px4.io/en/log/logging.html#configuration ? If so, could you log the default topics by removing the `logger_topics.txt` file on the SD card and post a log of a landing attempt with a full log?

Andree512, 2018-07-26:
[log2.zip](https://github.com/PX4/Firmware/files/2231635/log2.zip)
@ndepal In logger_topics.txt I prescribed the following:
landing_target_pose, 50
vehicle_local_position, 50
irlock_report, 50
vehicle_status, 50
In the log "10_35_21" the drone landed with the command "commander mode auto: precland" as a result, landed one meter from the lighthouse, a drone landed in the log "10_37_08", switching between the modes "commander mode auto: precland" and "Position mode" landed drone just to the lighthouse. I think the drone misses the location of the lighthouse with different heights and does not align itself with it

ndepal, 2018-07-26:
In both logs the vehicle is trying to do a precision landing, but is not able to center itself above the beacon.
At time 3:11 in the first log you can see the log message that the horizontal approach took too long. You see the same in the second log.

This is because the vehicle's position is not within `PLD_HACC_RAD` (0.2m) of the beacon. This is is likely due to poor position control performance. Have you tuned your vehicle to get tight control performance?
You can also try increasing `PLD_HACC_RAD` to make it easier for the vehicle to enter the descent phase of the precision landing. However, you will likely not get good accuracy if your position controller is not tuned well.

Andree512, 2018-07-26:
@ndepal 
Thank you! We see progress, but there is one more question: why does the drone search for the lighthouse only at the first stage of "Precision landing" at the altitude reduction stage, he does not continue to follow the lighthouse, and lands on the point where the lighthouse last saw. Is it possible to force the drone to move behind a beacon while decrease in altitude?

ndepal, 2018-07-26:
Under normal conditions, the vehicle tracks the beacon at all times. In your logs it does a fallback landing, i.e. it just lands at the current location.

The precision landing is composed of two main phases:
1. Horizontal approach: The vehicle holds its altitude and tries to center itself on the beacon
2. Descent above beacon: The vehicle descends and continues to try to keep the beacon centered

As I mentioned above, in your case the horizontal approach is never completed. The vehicle is unable to reduce it's position offset to the beacon enough to enter the second phase. In order to prevent the vehicle from just trying to approach horizontally until the battery runs out, there is a timeout to this phase.
When this timeout is reached, you see the message in the log `Precision landing took too long during horizontal approach` followed by `Falling back to normal land.` Now the vehicle simply descends where it is.

If you tune your vehicle such that the positional error can successfully be reduced to below `PLD_HACC_RAD` before the fallback timeout during the horizontal approach, you will see that it continues tracking the beacon as it descends.

Andree512, 2018-07-30:
@ndepal Hello!

I want to show a video of how our drone flies and logs with the magnified value PLD_HACC_RAD to 1.5 meters.
By this link: https://drive.google.com/open?id=16YN2U6zT6v8vAQtb3eShH7E6TMRGUE-C

ndepal, 2018-07-30:
In the first log you appear to have some issues with tracking. In the subsequent ones tracking seems to work, but control performance appears to be bad. Again, it's difficult to tell with so little data in the logs.
Please post a log with the default topics + `irlock_report`.

roboflot, 2018-10-08:
@ndepal
Hello Nicolas!
It seems that you are the only irlock hero in px4 stack ) My Pixhawk 4 (FMUv_5) does not start communication with Pixy, I described the problem in [this issue](https://github.com/PX4/Firmware/issues/10210). It seems that some i2c driver or parameter is missing in the code. Thanks for your help.

closed 2019-02-04 by stale[bot]