PX4/PX4-Autopilot #9908: GPS config if coming from arduplane/arducopter ecosystem : GPS not recognize

macfly1202, 2018-07-15:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

**Describe problem solved by the proposed feature**
Coming from Arduplane/arducopter ecosystem, GPS eeprom is configured to use 115kbps (if save config has been activated in arduplane parameter). In thi scase, Px4 firmware is not able to correctly configure GPS it at startup.

**Describe your preferred solution**
It will be nice to avoid use of u-center to configure it back to 38kps. Perhaps a trial of 115kbps to configure it, after a 9600 bps  ?

LorenzMeier, 2018-07-16:
Can you try this again with current master (in QGC, developer build)? I've merged a fix yesterday which resets the stored config. This should eliminate it.

macfly1202, 2018-07-16:
I manually reconfigure GPS (Holy Micro M8N, firmware 3.01) to 115 kbps. Reinstall Firmware developper build and still no GPS connection (My GPS has fix, blue light flashing). 

Manually reconfigure speed with u-center to 9600 bps give GPS back to life in Qgroundcontrol OR changing GPS_UBX_DYNMODEL to another setting than the actual (airborne with <4G acceleration) and reboot give GPS back in Qgroundcontrol. 

For now, your fix in dev build doesn't seem to work 'plug and play'. Thanks !

LorenzMeier, 2018-07-16:
Thanks - then we'll need to have another pass.

closed 2019-02-04 by stale[bot]