PX4/PX4-Autopilot #9920: SITL logger compile error with GCC 7.3

MaEtUgR, 2018-07-16:
**Bug**
I updated Cygwin to the newest version and it's using GCC 7.3
```
$ gcc --version
gcc (GCC) 7.3.0
```
When I compile SITL by `make posix` I get the following errors which weren't there before on this branch:
```
../../src/modules/logger/logger.cpp:2136:27: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
       statfs_buf.f_bavail >= (px4_statfs_buf_f_bavail_t)(min_free_bytes / statfs_buf.f_bsize)) {
       ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

../../src/modules/logger/logger.cpp:2172:26: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
  if (statfs_buf.f_bavail < (px4_statfs_buf_f_bavail_t)(50 * 1024 * 1024 / statfs_buf.f_bsize)) {
      ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

**Workaround**
Of course I can just add a `(int)` cast in front of the second operand of the comparison and then it compiles again but I'm writing this issue because there's likely some inconsistency with the types.

Why is `px4_statfs_buf_f_bavail_t` defined once signed, once usigned here? https://github.com/PX4/Firmware/blob/9ce83f22087311df06078edf3a9da640d93ab396/src/platforms/px4_defines.h#L115-L173

bkueng, 2018-07-16:
NuttX uses a different sign from Linux, and apparently Cygwin.

We should something like this (the type is only used in logger):
```
#ifdef __cplusplus
#include <sys/statfs.h>
typedef decltype(statfs::f_bavail) px4_statfs_buf_f_bavail_t;
#endif
```

MaEtUgR, 2018-07-17:
@bkueng Thanks for your suggestion! What puzzled me most is that this only happens after the last Cygwin update. I'll try/investigate such a solution and create a pr as soon as I found something reasonable.

PR #10009 (2018-07-24): Fix logger signed unsigned error with GCC 7.3 in Cygwin
Thanks to @bkueng for suggesting this change in #9920 !

closes #9920
closed 2018-07-24 by dagar