PX4/PX4-Autopilot #9923: python import error: ImportError('No module named toml',)

zouboan, 2018-07-16:
I'm using Ubuntu 14.04LTS

    I tried to compile v1.8.0:
    sudo make px4fmu-v3_default
    Got the error:
('python import error: ', ImportError('No module named toml',))

Required python packages not installed.

On a GNU/Linux or MacOS system please run:
  sudo pip install numpy toml

On Windows please run:
  easy_install numpy toml

 I tried installing with both pip and with apt-get and it installed properly. But, whenever I run the make of PX4 I get the same error.

hamishwillee, 2018-07-16:
16.04 LTS is the version of Ubuntu all our testing is done on. 14.04 might work, but it isn't "supported".

I would try installation using our convenience scripts: http://dev.px4.io/en/setup/dev_env_linux.html

They install toml as part of this:
```
sudo apt-get install python-argparse python-empy python-toml python-numpy python-dev python-pip -y
```

MaEtUgR, 2018-07-27:
@zouboan I'm not sure if that is related to your problem but when I was working on the Cygwin Toolchain to support PX4 development under Windows I had the exact same error happening when building. I followed down where this come from: https://github.com/PX4/Firmware/blob/b7d7d1209b331a571b205d5a0006e543c1891357/src/lib/mixer/geometries/tools/px_generate_mixers.py#L41-L53 and tried to manually import `toml` and `numpy` by manually executing the following commands on the bash shell:
```
python
import toml
import numpy
exit()
```
I found out that both packages were in fact installed but `numpy` failed to import whenever I ran the cygwin bash shell without the `--login` parameter (remember I'm on Windows). I honestly have no clue why this plays a role or if any of that translates to Ubuntu but you may try to look into the "login shell fails to import numpy" direction as well.

MaEtUgR, 2019-01-21:
I'm assuming this was solved. I could try the latest toolchain setup convenience scripts in a 14.04 VM but if there's no demand I won't go through that effort.

closed 2019-01-21 by MaEtUgR

machenxiang, 2019-08-07:
I have a new way to solve the problem.
First,      git clone https://github.com/uiri/toml.git 
Second,        cd toml
Third,           python setup.py install

jlm345, 2019-10-02:
I have got the same error when 
`make posix_sitl_default gazebo`
I solved it by modifying the PYTHONPATH and solve it:
`export PYTHONPATH=/usr/bin/python`
Hope it helps!

MaEtUgR, 2019-10-16:
> I honestly have no clue why this plays a role or if any of that translates to Ubuntu but you may try to look into the "non-login shell fails to import numpy" direction as well.

**Update:** By now I learned that it's not good to install python packages by running pip as root from https://dev.to/elabftw/stop-using-sudo-pip-install-52mn For a lot of packages it will result in a library installation that is tied to the root account and you can't import them using a different user.
So I think the solution is to run `pip install --user toml` and afterwards check again if you can import it as your user and it should work just fine.

liangpu597, 2020-05-17:
I have same troubles.  My system is : ubuntu 16.04 + kinetic + gazebo 9 
when I open a shell:
> python 
> import toml

and it works.
but i use : 
`make px4_sitl_default gazebo`
it errors:

> make px4_sitl_default gazebo
> -- PX4 version: v1.11.0-beta1
> -- PX4 config file: /home/tony/PX4_Firmware/boards/px4/sitl/default.cmake
> -- PX4 config: px4_sitl_default
> -- PX4 platform: posix
> -- PX4 lockstep: enabled
> -- cmake build type: RelWithDebInfo
> -- The CXX compiler identification is GNU 5.4.0
> -- The C compiler identification is GNU 5.4.0
> -- The ASM compiler identification is GNU
> -- Found assembler: /usr/bin/cc
> -- Check for working CXX compiler using: Ninja
> -- Check for working CXX compiler using: Ninja -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> -- Check for working C compiler using: Ninja
> -- Check for working C compiler using: Ninja -- works
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Detecting C compile features
> -- Detecting C compile features - done
> -- Building for code coverage
> -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
> -- build type is RelWithDebInfo
> -- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-255-g47624a0
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/tony/PX4_Firmware/build/px4_sitl_default
> [6/740] Generating mixer_multirotor_normalized.generated.h
> FAILED: cd /home/tony/PX4_Firmware/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py --normalize -f /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_pusher.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_y.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/tri_y.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
> Failed to import toml: No module named 'toml'
> 
> You may need to install it using:
>     pip3 install --user toml
> 
> [6/740] Generating mixer_multirotor.generated.h
> FAILED: cd /home/tony/PX4_Firmware/build/px4_sitl_default/src/lib/mixer/MultirotorMixer && /usr/bin/python3 /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/tools/px_generate_mixers.py -f /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_bottom_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/dodeca_top_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_plus.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_t.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/hex_x.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_cox_wide.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_plus.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/octa_x.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_deadcat.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_h.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_plus.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_s250aq.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_vtail.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_wide.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x_cw.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/quad_x.toml /home/tony/PX4_Firmware/src/lib/mixer/MultirotorMixer/geometries/qua [... cut ...]
> Failed to import em: No module named 'em'
> 
> You may need to install it using:
>     pip3 install --user empy
> 
> [6/740] Generating git version header
> ninja: build stopped: subcommand failed.
> Makefile:198: recipe for target 'px4_sitl_default' failed
> make: *** [px4_sitl_default] Error 1
> 

I think the means is told me : empy isn't installed.
and I use : `pip3 install --user toml` or: `pip install -user toml `
and try again: `make px4_sitl_default gazebo`

same errors  still exist.
who can tell me how to solve it ?
thank you

liangpu597, 2020-05-17:
> I have got the same error when
> `make posix_sitl_default gazebo`
> I solved it by modifying the PYTHONPATH and solve it:
> `export PYTHONPATH=/usr/bin/python`
> Hope it helps!

hi, I env found : PYTHONPATH is : `PYTHONPATH=/opt/ros/kinetic/lib/python2.7/dist-packages`
and how to change ?
thank you

jlm345, 2020-05-17:
> > I have got the same error when
> > `make posix_sitl_default gazebo`
> > I solved it by modifying the PYTHONPATH and solve it:
> > `export PYTHONPATH=/usr/bin/python`
> > Hope it helps!
> 
> hi, I env found : PYTHONPATH is : `PYTHONPATH=/opt/ros/kinetic/lib/python2.7/dist-packages`
> and how to change ?
> thank you
mabe
`export PYTHONPATH=/usr/bin/python`
if you want to keep the ROS path,you can try this one:
`export PYTHONPATH=/usr/bin/python:/opt/ros/kinetic/lib/python2.7/dist-packages`

liangpu597, 2020-05-17:
> I have got the same error when
> `make posix_sitl_default gazebo`
> I solved it by modifying the PYTHONPATH and solve it:
> `export PYTHONPATH=/usr/bin/python`
> Hope it helps!
>
>
> hi, I env found : [...]

I open a shell, and "export PYTHONPATH=/usr/bin/python:/opt/ros/kinetic/lib/python2.7/dist-packages"
and don't close the shell, continue :make px4_sitl_default gazebo
and same errors...........
please tell me , please...

jlm345, 2020-05-17:
> I open a shell, and "export PYTHONPATH=/usr/bin/python:/opt/ros/kinetic/lib/python2.7/dist-packages"
> and don't close the shell, continue :make px4_sitl_default gazebo
> and same errors........... [...]

You may try this.
Inquire the pythonPath Result, and set PYTHONPATH accurately.
`pip3 --version`
My Result:
`pip 19.3.1 from /usr/local/lib/python3.5/dist-packages/pip (python 3.5)`
Then set PYTHONPATH as this
`export PYTHONPATH=/usr/local/lib/python3.5/dist-packages:${PYTHONPATH}`
Then pip install packages for "No module named" should work!

liangpu597, 2020-05-17:
@jlm345 
> You may try this.
> Inquire the pythonPath Result, and set PYTHONPATH accurately.
> `pip3 --version`
> My Result:
> `pip 19.3.1 from /usr/local/lib/python3.5/dist-packages/pip (python 3.5)`
> Then set [...]

hi, i use : 
> pip3 --version

My result: pip 8.1.1 from /usr/lib/python3/dist-packages (python 3.5)
and i open a shell, set : export PYTHONPATH=/usr/lib/python3/dist-packages:${PYTHONPATH}
and i continue, don't close this shell, 

> make px4_sitl_default gazebo

and  same errors:

> Failed to import em: No module named 'em'

> You may need to install it using:

  

>   pip3 install --user empy

and i use: `pip3  install --user empy`
 
and errors:

> Exception:
> Traceback (most recent call last):
>   File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 555, in urlopen
>     self._prepare_proxy(conn)
>   File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 753, in _prepare_proxy
>     conn.connect()
>   File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/connection.py", line 252, in connect
>     ssl_version=resolved_ssl_version)
>   File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/util/ssl_.py", line 305, in ssl_wrap_socket
>     return context.wrap_socket(sock, server_hostname=server_hostname)
>   File "/usr/lib/python3.5/ssl.py", line 377, in wrap_socket
>     _context=self)
>   File "/usr/lib/python3.5/ssl.py", line 752, in __init__
>     self.do_handshake()
>   File "/usr/lib/python3.5/ssl.py", line 988, in do_handshake
>     self._sslobj.do_handshake()
>   File "/usr/lib/python3.5/ssl.py", line 633, in do_handshake
>     self._sslobj.do_handshake()
> socket.timeout: _ssl.c:629: The handshake operation timed out
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 209, in main
>     status = self.run(options, args)
>   File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 328, in run
>     wb.build(autobuilding=True)
>   File "/usr/lib/python3/dist-packages/pip/wheel.py", line 748, in build
>     self.requirement_set.prepare_files(self.finder)
>   File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 360, in prepare_files
>     ignore_dependencies=self.ignore_dependencies))
>   File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 577, in _prepare_file
>     session=self.session, hashes=hashes)
>   File "/usr/lib/python3/dist-packages/pip/download.py", line 810, in unpack_url
>     hashes=hashes
>   File "/usr/lib/python3/dist-packages/pip/download.py", line 649, in unpack_http_url
>     hashes)
>   File "/usr/lib/python3/dist-packages/pip/download.py", line 842, in _download_http_url
>     stream=True,
>   File "/usr/share/python-wheels/requests-2.9.1-py2.py3-none-any.whl/requests/sessions.py", line 480, in get
>     return self.request('GET', url, **kwargs)
>   File "/usr/lib/python3/dist-packages/pip/download.py", line 378, in request
>     return super(PipSession, self).request(method, url, *args, **kwargs)
>   File "/usr/share/python-wheels/requests-2.9.1-py2.py3-none-any.whl/requests/sessions.py", line 468, in request
>     resp = self.send(prep, **send_kwargs)
>   File "/usr/share/python-wheels/requests-2.9.1-py2.py3-none-any.whl/requests/sessions.py", line 576, in send
>     r = adapter.send(request, **kwargs)
>   File "/usr/share/python-wheels/CacheControl-0.11.5-py2.py3-none-any.whl/cachecontrol/adapter.py", line 46, in send
>     resp = super(CacheControlAdapter, self).send(request, **kw)
>   File "/usr/share/python-wheels/requests-2.9.1-py2.py3-none-any.whl/requests/adapters.py", line 376, in send
>     timeout=timeout
>   File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 610, in urlopen
>     _stacktrace=sys.exc_info()[2])
>   File "/usr/share/python-wheels/urllib3-1.13.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 228, in increment
>     total -= 1
> TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
> You are using pip version 8.1.1, however version 20.1 is available.
> You should consider upgrading via the 'pip install --upgrade pip' command.
>

julianoes, 2020-05-18:
Ok, and you have done?
```
pip3 install --user toml
```

If not, please paste the output of:
```
which python
which python3
which pip
which pip3
python --version
python3 --version
pip --version
pip3 --version
```

Princu1999, 2021-06-01:
pip3 install --user toml
I did the above command still it shows the same error .

$ which python
/usr/bin/python
$ which python3
/usr/bin/python3
$ which pip
/home/princu/.local/bin/pip
$ which pip3
/home/princu/.local/bin/pip3
$ python --version
Python 2.7.18
$ python3 --version
Python 3.8.5
$ pip --version
pip 21.1.2 from /home/princu/.local/lib/python3.8/site-packages/pip (python 3.8)
$ pip3 --version
pip 21.1.2 from /home/princu/.local/lib/python3.8/site-packages/pip (python 3.8)

> Ok, and you have done?
> 
> ```
> pip3 install --user toml
> ```
> 
> If not, please paste the output of:
> 
> ```
> which python
> which python3
> which pip
> which pip3
> python --version
> python3 --version
> pip --version
> pip3 --version
> ```