PX4/PX4-Autopilot #9931: FWIW: Naming conventions and styles

mcsauder, 2018-07-16:
Hi there,

I know that within PX4 astyle already enforces a lot of the code formatting, etc., but I hoped it might be useful if I posted a few bullet points for naming/indenting/etc that has helped my teams/projects align our styles during projects in the past.  Again, I hope this is useful to anyone coming across this post!

Coding standards:
- ISO standard C++ only
- C++11
- Use Doxygen style comments (with @ for commands) for every:
    - Class
    - Data member
    - Function
        - @brief command for every function
        - @param command (if not void) for every parameter
        - @return command (if not void)
- File
    - @copyright command

- Allman (BSD) indentation style

Naming Convention:
- Use CamelCase for everything: classes, variables, filenames.
- Classes start with a capitalized letter and are nouns.
- Variables start with a lowercase letter and are nouns.
- Routines start with a lowercase letter and begin with verbs. To name a function, use a description of the return value. To name a procedure, use a strong verb followed by an object. See Code Complete 2 §7.6 for the difference between a function and a procedure verb.
- Filenames are the name of the class or namespace within.  One class per file.
- Constants are named identically to variables.
- Globals are named identically to variables.
- Member variables are named identically to variables.

Indentation:

- Indentation shall be 4 space characters, not tabs.
- Braces shall begin and end on the indentation level.
- Namespaces are NOT indented.
- Case statements are NOT indented.
- Class visibility statements are NOT indented (public, protected, private)
- One statement per line.  This includes variable declarations.
- Do not put short if() ...; statements on one line.
- If the constructor initializers don't fit on a single line, put constructor initializer list items one per line, starting with the comma and aligned with the colon separator.  For example: 
    Class::Class()
      : var1(1)
      , var2(2)
    {
    ...
    }

(The purpose of this indentation policy, which can feel "incorrect" at times is to ensure that changes are isolated to the minimum number of lines.  Our tools: compilers, editors, diff viewers, and source code repository (git), all operate on a line-by-line basis.  When someone makes a change that affects a portion anywhere in the line, the tools consider the entire line changed.  This can lead to nasty issues like complex merge conflicts, or worse obscure the developer activity.)

Include Headers
A good practice is to include headers in the order most local to least local and alphabetized to whatever degree possible to avoid duplications. The purpose for this is to ensure that a proper dependency chain is maintained, because as projects grow, dependency driven compilation failures sometimes can be difficult to identify and resolve.

This means that header files have includes (alphabetized) in the order:
project includes
project dependency includes
system includes

Source files have the includes (alphabetized) in the order:
definition include
project includes
project dependency includes
system includes

I hope this might be useful in the future to anyone reading this!

-Mark

mcsauder, 2018-07-16:
@hamishwillee and anyone else, please feel free to close this issue anytime you feel it has served its' purpose!

hamishwillee, 2019-01-13:
Hi @mcsauder I'd hoped there would be more active discussion on this. 
What I've done is created https://github.com/PX4/Devguide/pull/717 that captures what I believe to be the information from this that we didn't have implicitly from using astyle. It isn't as detailed, but better reflects the actual rules that most developers adhere too. Hopefully having a PR will allow this to be discussed in more detail. 

Thanks for pushing this.