PX4/PX4-Autopilot #9939: Intel Aero altitude drops

santiago3dr, 2018-07-17:
Received new Intel Aero RTF and have issues maintaining altitude in ALT, POS and Auto running current master

Default parameters:
manual flight:
https://logs.px4.io/plot_app?log=bdb2de01-a8af-484c-9399-eb8e486e8e7d
auto take-off and rtl:
https://logs.px4.io/plot_app?log=39b0cf9c-1131-4b01-a8ea-e8faa23944b8

Increased EKF2_ACC_NOISE, EKF2_BARO_NOISE and EKF2_GYR_NOISE to double the default values respectively; minor improvement on the altitude drops but still seeing dips and corrections between most waypoints:
https://logs.px4.io/plot_app?log=bbc94f3a-331d-4554-9321-569a69f8d5ee
https://logs.px4.io/plot_app?log=30c1cae6-8180-4c20-b442-fe7dd305d25f

santiago3dr, 2018-08-01:
will troubleshoot and tune vehicle once issue #9984 has been addressed

dagar, 2018-08-22:
You had these issue with both Intel Aeros?

santiago3dr, 2018-08-22:
yes, the last one experienced the same altitude drops running both default parameters and increased EKF noise parameters

closed 2019-02-04 by stale[bot]