PX4/PX4-Autopilot #9944: How to build old version of Firmware

chmi0611, 2018-07-18:
Hi, I'm sorry to ask you a question not reporting a bug.

I'm dealing with Airsim by using PX4 and since PX4 v1.8.0 is not working for me, I try to use v1.6.3.

Because I'm not familiar with using git, it is hard for me to build old version of firmware.

I searched for someone who had similar problem with me and I found it (#9455).

But it does not help me a lot.

Can you tell me the procedure to build the old version of firmware step by step?

TSC21, 2018-07-18:
The comment https://github.com/PX4/Firmware/issues/9455#issuecomment-389955187 should have all you need.

closed 2018-07-18 by TSC21

hamishwillee, 2018-07-19:
And I have [updated](https://github.com/PX4/Devguide/pull/558) the build instructions to add a tip on this topic: 
http://dev.px4.io/en/setup/building_px4.html#get_px4_code