PX4/PX4-Autopilot #9949: No timestamp for SMBUS battery_status topic

AuterionWrikeBot, 2018-07-18:
Hi all,
The time stamp on this line would be reset to zero after voltage is updated.

https://github.com/PX4/Firmware/blob/00a47ba542b23f112e2e467a8fd5c3a1c8c322a9/src/drivers/batt_smbus/batt_smbus.cpp#L644-L651

I also doubt about battery_status from smbus is lower rate than ADC reading and not properly sent to mavlink.

AuterionWrikeBot, 2018-07-18:
➤ Daniel Agar commented:

Would you like to submit a PR fixing it? You could zero the struct above (https://github.com/PX4/Firmware/blob/00a47ba542b23f112e2e467a8fd5c3a1c8c322a9/src/drivers/batt_smbus/batt_smbus.cpp#L641) and drop the memset. Could you expand on the problem with ADC reading? If you make the timestamp fix and generate a simple log it might help.

dakejahl, 2018-07-18:
Please see PR #9620

closed 2018-07-18 by LorenzMeier

AuterionWrikeBot, 2018-09-27:
➤ Daniel Agar commented:

This has been fixed.