PX4/PX4-Autopilot #9951: Conflicting instructions for temperature calibration

AuterionWrikeBot, 2018-07-18:
The Parameter reference notes say to use SDLOG_PROFILE = 2 for "thermal calibration"
https://dev.px4.io/en/advanced/parameter_reference.html#sd-logging

However, the notes in the Python script and the User Guide say to use SDLOG_PROFILE = 4
https://github.com/PX4/Firmware/blob/b2046ffd6b68a0cd0a0e2226957a6b866e211967/Tools/process_sensor_caldata.py#L19
and
https://github.com/PX4/px4_user_guide/blob/40b09b378601038100094b37a0d78a3a9bb0e334/en/advanced_config/sensor_thermal_calibration.md

I assume SDLOG_PROFILE = 4 is correct for thermal calibration?

r1b4z01d, 2018-08-16:
Have either of you be able to successfully completed the calibration?

priseborough, 2018-08-17:
It's a bitmask, so the parameter needs to be set to 4 to toggle bit position 2 to true. The parameter description in code is correct.

closed 2019-03-27 by thomasgubler