PX4/PX4-Autopilot #9964: HITL review

AuterionWrikeBot, 2018-07-18:
* what currently works
  * jmavsim
  * gazebo
  * xplane
  * flightgear
* what do we want to support
* can we make commander oblivious to HIL (commander rewrite)
* should we consider unifying with the SITL interface (flash considerations aside)

AuterionWrikeBot, 2018-07-18:
➤ Nuno Marques commented:

@arwagoner maybe you can leave here also your latest experience with the Gazebo side and the current state

AuterionWrikeBot, 2018-07-18:
➤ arwagoner commented:

In my experience, only quadcopters are reliably supported in Gazebo. The standard vtol airframe is able to takeoff in Gazebo, but it doesn't seem to respond to any landing commands. Neither plane airframe (sys_autostart 1000, 2100) works with Gazebo at all. If it is feasible, it would be great to have all types of airframes (at least their generic versions) working with one simulator.

AuterionWrikeBot, 2018-07-18:
➤ Hamish Willee commented:

Re "what currently works", the docs list is here: https://dev.px4.io/en/simulation/hitl.html#compatible_airframe * There is no documentation of flightgear, so I'd argue that even if it doesn't currently work even if if in theory it might. * The [xplane images for settings](https://dev.px4.io/en/simulation/hitl.html#using-x-plane-fixed-wing-only) are for an older version. I got this kind of flying specifying a "Cessna model". It wasn't clear to me whether this actually represents the PX4 UAV or not. Upshot, I'd say "not clear if actually works". * There is no VTOL HITL support. I don't know what we want to support, but I'd argue that if we're going to support something: * It should be supported, tested, and known to work - not "ambiguously so" like Flightgear and Xplane * We should support at least one of each frame type (vtol, plane, quadcopter) in at least one simulator. * The setup in QGC should be the same in all cases. I like the idea of having a simulator that supports all vehicle types, and gazebo would seem to be the one with the greatest traction.

AuterionWrikeBot, 2018-07-18:
➤ Jasmine commented:

I am doing research on HITL platform of Tailsitter. I'm very interested on this function. Could I join this?

AuterionWrikeBot, 2018-07-18:
➤ Hamish Willee commented:

@jingego Sure, but probably better to hold actual "discussion" on slack in #hitl channel.

julianoes, 2019-07-31:
This is a confusing issue, I'm going to close it.

closed 2019-07-31 by julianoes

julianoes, 2019-07-31:
Come on, I won against the bot!

hamishwillee, 2019-07-31:
@julianoes It's not over until you lock the issue in a way that the bot cannot unlock it :-)