PX4/PX4-Autopilot #9965: VTOL: ATTENTION!!! DANGER

AuterionWrikeBot, 2018-07-18:
PUSHER MOTOR STARTS DURING AIRSPEED CALIBRATION
on the bench.

The plane quad delta wing, the tubes from the pitot were swapped.

We recreated it two times

The second time by any chance we had logging enabled all the time so we have it on log.

https://logs.px4.io/plot_app?log=de51dc5e-e868-426f-90c3-1fcc42855fe1

It happened after we blew in the pitot at the end of the calibration. The calibration went wrong (because of the reversed tubes). Both times the motor started.

After we corrected the tubes never happened again.

AuterionWrikeBot, 2018-07-18:
➤ Daniel Agar commented:

I don't see current in your log, but you can tell from the gyro where it started. [image]

AuterionWrikeBot, 2018-07-18:
➤ Daniel Agar commented:

[image]

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

There is no actuator controll! But there are 1100us on the servo bus... It is Idle as RPM but still could hurt you. (Out motor is at main ch7) [image]

AuterionWrikeBot, 2018-07-18:
➤ Daniel Agar commented:

Can you run pwm info and share the output?

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

Will try I'm not at the location...

AuterionWrikeBot, 2018-07-18:
➤ Daniel Agar commented:

Questions 1. why did output 6 go to 1100us? 2. why didn't the px4io keep output 6 disarmed? @tubeme it possible please capture pwm info in a few different states. After a fresh boot, armed in MC mode, armed in FW mode.

AuterionWrikeBot, 2018-07-18:
➤ Sander Smeets commented:

@tubeme why does your log only contain 30% of the params?

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

@sanderux Really don't know...

AuterionWrikeBot, 2018-07-18:
➤ Sander Smeets commented:

Can you send the param file for your airframe?

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

@sanderux Sorry I was away for a while. I will get the param file when the person comes back from vacation. Other then this now the plane does not make this problem. May it was because of the reversed tubes from the pitot... We will try to recreate again.

AuterionWrikeBot, 2018-07-18:
➤ Roman Bapst commented:

@tubeme Did the person come back from his vacation?

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

Yes here is the parameter dump
[Parameter VTL3X 2018_Feb_6 (1.6.5).zip](https://github.com/PX4/Firmware/files/1748151/Parameter.VTL3X.2018_Feb_6.1.6.5.zip)

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

It continues to do it all the time. After calibration for example, every time. It switches the pusher to 1100us.

AuterionWrikeBot, 2018-07-18:
➤ Roman Bapst commented:

@tubeme I was unable to unzip the file you attached. Can you verify that it's a real zip file?

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

Hmmm I use rarlab but I told it to make zip. Probably something went wrong. Can you try a RAR instead... BTW since we reported this issue, we can reproduce it each and everytime with AUAV X2 + 1.6.5 after calibration. Does not matter what kind of calibration. It does it to the mag cal to the gyro cal or acc cal. The worst thing we have to disconnect the battery to stop it. The arm button does not affect it.

AuterionWrikeBot, 2018-07-18:
➤ Vasil Petrov commented:

@RomanBapst Did you manage to open the file or do you need me to re zip it?

julianoes, 2019-07-09:
@RomanBapst what's going on here?

RomanBapst, 2019-07-10:
We are not able to see the log as it's private and does not exist anymore.
Furthermore, we have been doing many airspeed calibrations and have never encountered this problem.

closed 2019-07-10 by RomanBapst