PX4/PX4-Autopilot #9971: Snapdragon built broken in master

r0gelion, 2018-07-18:
Current master (3293fe4) is broken for Snapdragon (eagle_default), the ESCs are not initializing with this commit.

Last working commit is: c199ca5

Commit 2931b85 is giving an error and it's not compiling
[ 94%] Building CXX object src/modules/commander/CMakeFiles/modules__commander.dir/state_machine_helper.cpp.o
[ 94%] Linking CXX static library libecl_EKF.a
[ 94%] Built target ecl_EKF
[ 94%] Linking CXX static library libmodules__commander.a
[ 94%] Built target modules__commander
make[1]: *** [all] Error 2
make: *** [qurt_eagle_default] Error 2

@dagar

santiago3dr, 2018-07-24:
flashed current master 80cac65
same behavior, esc's non responsive

dagar, 2018-09-01:
[image]

karliss, 2018-12-05:
I was able to get it somewhat working, will probably send a pull request tomorrow after some cleanup and testing with latest master. Problems where mostly caused by dynamic linker errors on DSP
* inconsistent extern "C" usage, probably caused by #10267 or other similar refactoring - easy to fix
* shmem_qurt param_find_changed, it is difficult to tell if my fix is correct without understanding how the code is supposed to work
* _Stof - added one more function to qurt_stubs see also bsearch  bellow
* bsearch - This one is bit weird as other libc functions are available. It is not difficult to reimplement it but it would be better to figure out how libc linking is supposed to work for fastrpc modules.

karliss, 2018-12-05:
One more thing I forgot to mention was differences in generated parameter list. Possix half had a few more parameters than qurt causing offset when synchronizing paremeters between them.

dagar, 2018-12-05:
@karliss it would be great to get this working again. We now have a snapdragon flight connected to the build system, so once we get it working again we'll be able to keep it working easily. Please ping me on PX4 slack if you have any questions.

dagar, 2018-12-05:
I can probably figure out the param sync issue. I don't understand bsearch though.

dagar, 2018-12-05:
The parameter mismatch is due to the new serial port configuration.

	    <parameter default="1" name="MAV_0_FORWARD" type="INT32">
	      <short_desc>Enable MAVLink Message forwarding for instance 0</short_desc>
	      <long_desc>If enabled, forward incoming MAVLink messages to other MAVLink ports if the message is either broadcast or the target is not the autopilot. This allows for example a GCS to talk to a camera that is connected to the autopilot via MAVLink (on a different link than the GCS).</long_desc>
	      <boolean />
	      <reboot_required>True</reboot_required>
	    </parameter>
	    <parameter default="0" name="MAV_0_MODE" type="INT32">
	      <short_desc>MAVLink Mode for instance 0</short_desc>
	      <long_desc>The MAVLink Mode defines the set of streamed messages (for example the vehicle's attitude) and their sending rates.</long_desc>
	      <reboot_required>True</reboot_required>
	      <values>
		<value code="0">Normal</value>
		<value code="1">Custom</value>
		<value code="2">Onboard</value>
		<value code="3">OSD</value>
		<value code="4">Magic</value>
		<value code="5">Config</value>
		<value code="7">Minimal</value>
	      </values>
	    </parameter>
	    <parameter default="1200" name="MAV_0_RATE" type="INT32">
	      <short_desc>Maximum MAVLink sending rate for instance 0</short_desc>
	      <long_desc>Configure the maximum sending rate for the MAVLink streams in Bytes/sec. If the configured streams exceed the maximum rate, the sending rate of each stream is automatically decreased. If this is set to 0, a value of &lt;baudrate&gt;/20 is used, which corresponds to half of the theoretical maximum bandwidth.</long_desc>
	      <min>0</min>
	      <unit>B/s</unit>
	      <reboot_required>True</reboot_required>
	    </parameter>
	    <parameter default="0" name="MAV_1_FORWARD" type="INT32">
	      <short_desc>Enable MAVLink Message forwarding for instance 1</short_desc>
	      <long_desc>If enabled, forward incoming MAVLink messages to other MAVLink ports if the message is either broadcast or the target is not the autopilot. This allows for example a GCS to talk to a camera that is connected to the autopilot via MAVLink (on a different link than the GCS).</long_desc>
	      <boolean />
	      <reboot_required>True</reboot_required>
	    </parameter>
	    <parameter default="2" name="MAV_1_MODE" type="INT32">
	      <short_desc>MAVLink Mode for instance 1</short_desc>
	      <long_desc>The MAVLink Mode defines the set of streamed messages (for example the vehicle's attitude) and their sending rates.</long_desc>
	      <reboot_required>True</reboot_required>
	      <values>
		<value code="0">Normal</value>
		<value code="1">Custom</value>
		<value code="2">Onboard</value>
		<value code="3">OSD</value>
		<value code="4">Magic</value>
		<value code="5">Config</value>
		<value code="7">Minimal</value>
	      </values>
	    </parameter>
	    <parameter default="0" name="MAV_1_RATE" type="INT32">
	      <short_desc>Maximum MAVLink sending rate for instance 1</short_desc>
	      <long_desc>Configure the maximum sending rate for the MAVLink streams in Bytes/sec. If the configured streams exceed the maximum rate, the sending rate of each stream is automatically decreased. If this is set to 0, a value of &lt;baudrate&gt;/20 is used, which corresponds to half of the theoretical maximum bandwidth.</long_desc>
	      <min>0</min>
	      <unit>B/s</unit>
	      <reboot_required>True</reboot_required>
	    </parameter>
	    <parameter default="0" name="MAV_2_FORWARD" type="INT32">
	      <short_desc>Enable MAVLink Message forwarding for instance 2</short_desc>
	      <long_desc>If enabled, forward incoming MAVLink messages to other MAVLink ports if the message is either broadcast or the target is not the autopilot. This allows for example a GCS to talk to a camera that is connected to the autopilot via MAVLink (on a different link than the GCS).</long_desc>
	      <boolean />
	      <reboot_required>True</reboot_required>
	    </parameter>
	    <parameter default="0" name="MAV_2_MODE" type="INT32">
	      <short_desc>MAVLink Mode for instance 2</short_desc>
	      <long_desc>The MAVLink Mode defines the set of streamed messages (for example the vehicle's attitude) and their sending rates.</long_desc>
	      <reboot_required>True</reboot_required>
	      <values>
		<value code="0">Normal</value>
		<value code="1">Custom</value>
		<value code="2">Onboard</value>
		<value code="3">OSD</value>
		<value code="4">Magic</value>
		<value code="5">Config</value>
		<value code="7">Minimal</value>
	      </values>
	    </parameter>
	    <parameter default="0" name="MAV_2_RATE" type="INT32">
	      <short_desc>Maximum MAVLink sending rate for instance 2</short_desc>
	      <long_desc>Configure the maximum sending rate for the MAVLink streams in Bytes/sec. If the configured streams exceed the maximum rate, the sending rate of each stream is automatically decreased. If this is set to 0, a value of &lt;baudrate&gt;/20 is used, which corresponds to half of the theoretical maximum bandwidth.</long_desc>
	      <min>0</min>
	      <unit>B/s</unit>
	      <reboot_required>True</reboot_required>
	    </parameter>

karliss, 2018-12-05:
About the bsearch problem. From what I understood hexagon dsp code gets built as dynamic library which gets loaded by fastrpc module.  It links standard c library dynamically, probably to prevent conflicts between multiple dynamic libraries.  It seems that ADSP firmware image for which comes from Intrinsyc for sanpdragon flight boards doesn't include bsearch function. `adsp_info check_sym` also says that bsearch is not available.  I'd be happy to hear that I misunderstood something or that there is newer dsp firmware image available.

karliss, 2018-12-06:
Tested with latest master+my changes+#10992. Few more things broke during, probably during board layout reorganization. Some minor bugs in upload commands and DF_DRIVERS don't get properly built (they don't appear in `qshell help` output).

karliss, 2018-12-06:
My work in progress will be found here https://github.com/karliss/Firmware/tree/snapdragon_fix_build2

dagar, 2018-12-06:
@karliss great progress!

Feel free to open a WIP pull request on top of #10992. We also need to think of a very simple test the CI system can perform with the hardware attached. Build, upload, run, verify all expected modules and drivers are running on both sides.

santiago3dr, 2019-01-14:
flashed today's master
same behavior, esc's do not initialize

julianoes, 2019-01-29:
What's the output in `mini-dm` (or `nano-dm`) and in the shell?

julianoes, 2019-07-01:
@r0gelion @santiago3dr can you try again? This should have been fixed in the meantime.

dagar, 2019-07-01:
Snapdragon was most recently fixed (and tested) by @karliss. https://github.com/PX4/Firmware/pull/11000

closed 2019-07-01 by dagar