PX4/PX4-Autopilot #9974: MAV_CMD_DO_REPOSITION Ground Speed

kfruson, 2018-07-18:
Setting Param1 for command MAV_CMD_DO_REPOSITION is returning the default (-1). The pixhawk groundspeed needs to be set to inputted Param1 as the Mavlink message says it does.

hamishwillee, 2018-07-18:
Is the command supported? Quick check of mavlink_mission.cpp seems to indicate maybe not. Does it return the unsupported message error in SITL (Debug) ?https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L1482

kfruson, 2018-07-19:
Command is supported. PX4 moves to specified location when using the REPOSITION, tested with SITL. Only thing that is not working is the speed command. After quick look in the code it is returning the default -1. Just needs to be changed to use the cmd.param1 which is specified in the Mavlink Documentation.

hamishwillee, 2018-07-20:
Just FMI, where is this set in code?

kfruson, 2018-07-20:
For sure.

https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator_main.cpp

Around Line 320 is where it starts. 336 is where it looks like it gets the speed.

hamishwillee, 2019-01-03:
Looks like this might be fixed in https://github.com/PX4/Firmware/commit/21bcc0d2f40d307c3c88009ff5a7f13e8dbbe93e

kfruson, 2019-01-03:
Roger, closing as this appears to be fixed in the issue mentioned above.

closed 2019-01-03 by kfruson