PX4/PX4-Autopilot #9977: Aerotenna OcPoc failing to engage Auto in master

santiago3dr, 2018-07-18:
running master 3293fe4
successful flight with auto take-off and rtl as part of mission:
https://logs.px4.io/plot_app?log=593203b8-e65b-4a5a-9a9a-9cbd65ac5127

after landing, armed again but Auto would not engage
took off manually and still, Auto was not responsive:
https://logs.px4.io/plot_app?log=9fb47e82-36d9-4600-a868-c09b1df59bb0

swapped batteries and Auto still not responsing:
https://logs.px4.io/plot_app?log=0b6bd97b-721f-4acf-aacd-d7147bf9c73a

santiago3dr, 2018-07-30:
flashed current master ca42483
no change

andrewmegaris, 2018-07-31:
@santiago3dr 
I compiled a fresh PX4 binary from commit [13f2315](https://github.com/PX4/Firmware/commit/13f231531436afba3c709acbc63b4315011b75d4) today(July 31).  I was able to load a mission, engage auto then execute the mission.  After the mission was complete I was able to toggle to manual, then back to auto and execute the mission a second time.  Would you please attempt to recreate your original scenario using the linked commit and see if this has any affect?

santiago3dr, 2018-08-01:
flashed latest master, mission upload looks to be working, will test fly later in the day.

andrewmegaris, 2018-08-02:
I hope your flight went well!  Please do not forget to tag this issue if closed if you are satisfied.

santiago3dr, 2018-08-02:
Original issue happened again with some odd results
setup and execution:
flashed master daeb2d8
connected to qgc and loaded a flight plan
armed in stabilized and then engaged auto, successful flight:
https://logs.px4.io/plot_app?log=3d55c42a-00bd-44a4-8d07-ad0b4d442842

landed from previous flight and without a power cycle, armed in stabilized, switched to auto but qgc reported no valid mission:
https://logs.px4.io/plot_app?log=eef47b3d-5180-4347-b73d-86c29ae18912

Swapped batteries and without re-loading a mission, was able to execute mission successfully:
https://logs.px4.io/plot_app?log=0aff1b66-d387-4631-ad33-3ff840cc8333

landed from previous flight and without a power cycle, armed in stabilized, switched to auto.
successful take-off but copter idled at about 2 meter altitude without executing the mission
https://logs.px4.io/plot_app?log=5a3a391e-3aae-47b6-ad40-7fe7a37c55ce
qgc reported the vehicle was in Mission

Given the flight plan was not removed from the vehicle, this is only a minor inconvenience and I can simply power cycle between flights.
Thoughts?

JohnSnowball, 2018-08-09:
Hi, I got an additional question, this is my first time see people using Aerotenna OcPoc, could you please send me some references about this flight control unit? Which version do we use for this, fmu-v5 or something else? Thanks in advance.

andrewmegaris, 2018-08-10:
@santiago3dr Today(Aug -10) we rebuilt PX4 off commit 'daeb2d8'.  We where able to execute the same mission multiple times, adjust the mission, upload adjusted mission, and re-execute all in the field with no issues.  I would recommend clearing the SD card(personally I would re partition too). Then rebuild the firmware and load it all fresh.

@JohnSnowball [ PX4's OcPoC documentation ](https://docs.px4.io/en/flight_controller/ocpoc_zynq.html ),  OcPoC also has it's own [User Hub](https://aerotenna.readme.io/docs/ocpoc-mini-zynq-specifications).

santiago3dr, 2018-08-10:
i'll give it a shot

closed 2019-02-04 by stale[bot]