PX4/PX4-Autopilot #9984: Intel Aero: mission upload broken on master

santiago3dr, 2018-07-19:
Flashed master ade70cb and the mission that was previously stored seemed to be deleted.
Attempted to upload mission and got error: (MAV_MISSION_ERROR) & Unable to write on micro SD
[image]
[image]

lbegani, 2018-07-25:
@santiago3dr  Is it easily reproducible or was it seen only once?

From the log, it seems like navigator has entered some invalid state and hence not saving new mission.

```
Logged Messages

10:00:21WARNING[load_mon] tap_esc low on stack! (228 bytes left)
20:00:30WARNING[load_mon] tap_esc low on stack! (228 bytes left)
30:00:39WARNING[load_mon] tap_esc low on stack! (228 bytes left)
40:00:48WARNING[load_mon] tap_esc low on stack! (228 bytes left)
50:00:57WARNING[load_mon] tap_esc low on stack! (228 bytes left)
60:01:02WARNING[navigator] No valid mission available, refusing takeoff.
70:01:05WARNING[navigator] No valid mission available, loitering.
80:01:24WARNING[load_mon] tap_esc low on stack! (228 bytes left)
90:01:31WARNING[navigator] Invalid mission state.
100:01:31ERROR[navigator] Can't save mission state
110:01:33WARNING[load_mon] tap_esc low on stack! (228 bytes left)

santiago3dr, 2018-07-25:
I have attempted with daily releases of master and I'm getting this sane error.

santiago3dr, 2018-07-25:
Reverted to 1.8.0 stable; mission upload goes through successfully so this is definitely on the software size, I'll try to find the specific commit that broke said function.

santiago3dr, 2018-07-30:
flashed current master ca42483
no change

LorenzMeier, 2018-07-30:
@anitha-suresh Any chance someone from the team can have a look?

closed 2019-02-04 by stale[bot]