PX4/PX4-Autopilot #9986: Ubuntu 18.04 toolchain - Nuttx Virtual Floating Point issue

mcsauder, 2018-07-20:
Ubuntu 18.04LTS doesn't play well with the PX4/Nuttx code base.  The issue is with virtual floating point register arguments not agreeing with build instructions at the PX4 level.

This is not an issue with Ubuntu 16.04, only with the advancement of the compiler toolchains in 18.04LTS.
Here is the output of the first list of compilation errors, although it outputs hundreds of lines of the same errors: 
```
[100%] Linking CXX executable ../../nuttx_px4fmu-v4_default.elf
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: ../../nuttx_px4fmu-v4_default.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libm.a(lib_a-s_fpclassify.o) does not
```

I've tried fixing this by replacing soft floating point CMakeList.txt arguments throughout the project with hard floating point arguments, however, I have had no success in resolving the issue.

```mark@mark-laptop:~/code/Firmware$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.3.0-16ubuntu3' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --with-as=/usr/bin/x86_64-linux-gnu-as --with-ld=/usr/bin/x86_64-linux-gnu-ld --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3) 
```

nkhoit, 2018-07-24:
This works for me on 18.04LTS. It looks like you're trying to build with gcc-arm 6.3.1. 
I'm using 7-2017-q4-major 7.2.1 20170904 that's linked in http://dev.px4.io/en/setup/dev_env_linux_ubuntu.html in the NuttX section.

mcsauder, 2018-07-27:
@nkhoit , Thank you for your feedback!  I appreciate it very much.  Turns out that my issue was with installing the toolchain and [Ubuntu 18.04 secure boot/MOK](https://wiki.ubuntu.com/UEFI/SecureBoot).

The trouble with my install appears to have been that when I rebooted I did not acknowledge the MOK to allow the toolchain updates to be applied.  I don't know if this is a problem anyone else might encounter, but at least with my installation it was a challenge to sort out.  I wouldn't have resolved this without your reply.  Thank you very much!!!

-Mark

closed 2018-07-27 by mcsauder