PX4/PX4-Autopilot #9988: Gazebo simulation issue

pranavn248, 2018-07-20:
Hi,
The mode in QGC is in mission mode and it is able to takeoff (Even though it is slow).
I simulated my pixhawk in HITL mode with Gazebo and QGroundControl but the simulation is very slow. I'm getting a really low fps n a real time factor of 0.5. I've attached a screenshot of my simulation. 
https://i.imgur.com/9667AQf.png
Kindly help me overcome this issue.
Thanks in Advance :)

ndepal, 2018-07-20:
Try using a gazebo world with a flat ground. The terrain in your world is computationally expensive and slows down the simulation significantly depending on your computer's hardware.

pranavn248, 2018-07-20:
Sorry to not have mentioned earlier but I'm running it on a virtual machine so now I understand that could be the reason. Do you know how to improve the performance of my virtual machine?

ndepal, 2018-07-20:
Like I said, you can run a simpler world. You can also run gazebo without the 3D visualization by prepending `HEADLESS=1` to your `make` command.

VMs are generally not recommended for simulation because it tends to be resource intensive.

TSC21, 2018-12-10:
Closing, @pranavn248 please open if it still persists.

closed 2018-12-10 by TSC21