PX4/PX4-Autopilot #9989: not rc detected on master

dannyfpv, 2018-07-20:
On the pr and master the copter does not detect the radio, only on stable works well
first  send to manual mode and later to hold mode

https://review.px4.io/plot_app?log=bf71d883-0596-4723-95cc-ed70f2412ce6

Add screenshots to help explain your problem.

**Drone 
f450, pixhack, 20ams hobbywing esc and iris motors 

on the firmware master 3293fe still work good @dagar

Tony3dr, 2018-07-30:
@dagar correct this is for the Pixhack v5 works fine with Pixhawk 4.

dagar, 2018-07-30:
FYI @davids5

Tony3dr, 2018-07-31:
[image]
[image]
@davids5 We are having issues with the new master . With master 3293fe, it works fine. Attached are the Images.

closed 2019-01-08 by mhkabir

tops4u, 2019-01-12:
> The developer version of Pixhack v5 does not further support. We can close this issue.

Hi @jinger26 could you elaborate?? I have exactly the same Problem! RC is detected on 1.8.1 but not on 1.9.0 Alpha / Master. Is my v5 now a brick for PX4 Community?

jinger26, 2019-01-15:
Hi @tops4u , Pixhack v5 that has a "developer version" marked on the hardware was a release candidate for evaluation purpose only and not in production anymore. Do you have that version of the hardware or do you have a "V5" version of the hardware?

tops4u, 2019-01-15:
@jinger26 I have not seen such a marking on my HW. On the console it shows up as HW Ver V551.

davids5, 2019-01-15:
@tops4u It is marked on the top of the cube. In white lettering.  Based on the value I would assume you have V5. 

See https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v5/src/manifest.c#L88-L91 

Master supports the FMUv5 Spec. 1.8.1 did not have the full RC (PPM, SBUS) support for FMUv5. The  board you have is not wired to the FMUv5 Spec for RC. But as I say above it is unclear to me which chips, resistors are populated and what the we could get working **100% reliably**

davids5, 2019-01-15:
@tops4u 

>Shall I open a new Issue for my non working RC? 

It is a know issue. 

>Should I have a look inside? 

I do not think it will add value. We do not have a way to ID the components back to the schematic. The vendor has not given me a list of the pop option and how they are installed. So I can not tell you what has to be removed/fixed. 

The change that is most likely affecting the circuit is initializing the F7's USART 6 and not getting it it to "Single wire" mode.  You could test this By removing USART 6 from the boar/nuttx config on the FMU and ensuring those GPIOs are left as inputs.

> Why does Boot stop only due to Config change? 

I am not sure what you mean. 

>Should I open an Issue for this?

Yes and give full details

tops4u, 2019-01-15:
@davids5  could you tell me where the known Issue is tracked? I could not find an issue for this. 

I've opened Issue #11223 for the Boot issue. 

Regarding the RCIN Problem I have could this be a Problem with Communication to the IO? This is what I get when I send px4io Status to the v5:

```
> px4io status 
WARN  [px4io] loaded
protocol 2147483648 hardware 2147483648 bootloader 2147483648 buffer 2147483648B crc 0x8000000080000000
2147483648 controls 2147483648 actuators 2147483648 R/C inputs 2147483648 analog inputs 2147483648 relays
2147483648 bytes free
status 0x0000 SAFETY_SAFE RC_FAIL FMU_FAIL MIXER_FAIL ARM_NO_SYNC INIT_FAIL
alarms 0x0000
vservo 2147483648 mV vservo scale 2147483648
vrssi 2147483648
actuators 0 0 0 0 0 0 0 0
servos 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
0 raw R/C inputs
R/C flags: 0x0000
mapped R/C inputs 0x0000
ADC inputs
features 0x0000
arming 0x0000 FMU_DISARMED IO_ARM_DENIED
rates 0x80000000 default 2147483648 alt 2147483648 sbus 2147483648
debuglevel 2147483648
controls 0: 0 0 0 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
input 0 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 1 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 2 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 3 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 4 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 5 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 6 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 7 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 8 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 9 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 10 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 11 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 12 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 13 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 14 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 15 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 16 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
input 17 min 2147483648 center 2147483648 max 2147483648 deadzone 2147483648 assigned 2147483648 options 0x0000
failsafe 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
disarmed values 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648 2147483648
IMU heater off
```

Using Version:
```
HW arch: PX4_FMU_V5
HW type: V515
HW version: 0x00000001
HW revision: 0x00000005
FW git-hash: d8ab059ff3dcd50e0b8953da4de4150dffc2b75e
FW version: 1.9.0 40 (17367104)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: c225d4a691ef4572839b19635a84f2d9bf07ea74
Build datetime: Jan 15 2019 08:25:45
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0002000000003833393330385108001b0024
MCU: STM32F76xxx, rev. Z
[... 42 lines cut ...]
input 15 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 16 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 17 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
failsafe 0 0 0 0 0 0 0 0
disarmed values 0 0 0 0 0 0 0 0
```

```
Version is:
HW arch: PX4FMU_V5
HW type: V551
HW version: 0x00000001
HW revision: 0x00000005
FW git-hash: f13bbacd5277123d6af2cb5ed21587c220031353
FW version: Release 1.8.2 (17302271)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 63775322bf25adb406594f8e610122fe0cef2f7a
Build datetime: Jan 14 2019 12:41:43
Build uri: BUILD_URI
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
MFGUID: 3833393330385108001b0024
MCU: STM32F76xxx, rev. Z
UID: 1B0024:30385108:38333933 
```

davids5, 2019-01-15:
@tops4u 

Todays master tested on 2 units. Works on my HW.
I have a XSR using the SBUS pins connected to the RC inputs 
TOP
Ground
Power
Signal

```
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V5
HW type: V515
HW version: 0x00000001
HW revision: 0x00000005
FW git-hash: d8ab059ff3dcd50e0b8953da4de4150dffc2b75e
FW version: 1.9.0 40 (17367104)
FW git-branch: master
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: c225d4a691ef4572839b19635a84f2d9bf07ea74
Build datetime: Jan 15 2019 14:53:32
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 000200000000363539373137510100290018
MCU: STM32F76xxx, rev. Z
[hardfault_log] Fault Log is Armed
INFO  [tune_control] Publishing standard tune 1
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 350216 bytes
INFO  [param] selected parameter default file /fs/mtd_params
rgbled on I2C bus 1 at 0x55 (bus: 100 KHz, max: 100 KHz)
WARN  [rgbled_ncp5623c] no RGB led on bus #1
Board init: /etc/init.d/rc.board
MPU6000 on SPI bus 1 at 1 (1000 KHz)
WARN  [mpu6000] unexpected ID 0x00 accepted
MPU6000 on SPI bus 1 at 0 (1000 KHz)
WARN  [mpu6000] unexpected ID 0x00 accepted
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
INFO  [ist8310] no device on bus 1
INFO  [ist8310] no device on bus 2
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
ERROR [pmw3901] driver start failed
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
WARN  [bst] no devices found
INFO  [mpu9250] Bus probed: 2
WARN  [mpu9250] no device on bus 2
px4flow [141:100]
INFO  [px4flow] scanning I2C buses for device..
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [px4io] default PWM output device
INFO  [init] Mixer: /etc/mixers/octo_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
INFO  [logger] logger started (mode=all)

NuttShell (NSH)
nsh> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)

nsh> px4io status
WARN  [px4io] loaded
protocol 4 hardware 2 bootloader 3 buffer 64B crc 0xb19cbe0d
8 controls 8 actuators 18 R/C inputs 2 analog inputs 0 relays
1544 bytes free
status 0x0e24 SAFETY_SAFE RC_OK SBUS FMU_FAIL MIXER_FAIL ARM_SYNC INIT_OK FAILSAFE
alarms 0x0010 FMU_LOST
vservo 2 mV vservo scale 10000
vrssi 1007
actuators -25000 -25000 -25000 -25000 -25000 -25000 -25000 -25000
servos 0 0 0 0 0 0 0 0
reversed outputs: [________] trims: r:   0.0000 p:   0.0000 y:   0.0000
18 raw R/C inputs 1498 1496 1212 1490 1674 1499 982 982 1494 1494 1494 1494 1494 1494 1494 1494 998 998
R/C flags: 0x0010
mapped R/C inputs 0x0000
ADC inputs 4 1754
features 0x0008 RSSI_ADC
arming 0x0401 FMU_DISARMED IO_ARM_OK OVERRIDE_IMMEDIATE
rates 0x0000 default 50 alt 200 sbus 72
debuglevel 0
controls 0: 0 0 0 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
input 0 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 1 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 2 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 3 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 4 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 5 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 6 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 7 min 1000 center 1500 max 2000 deadzone 10 assigned 255 options 0x0000
input 8 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 9 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 10 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 11 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 12 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 13 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 14 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 15 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 16 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
input 17 min 1000 center 1500 max 2000 deadzone 0 assigned 255 options 0x0000
failsafe 0 0 0 0 0 0 0 0
disarmed values 0 0 0 0 0 0 0 0
```

davids5, 2019-01-15:
All I did was set the SYS_AUTOSTART to 4011

davids5, 2019-01-15:
I grabbed your params from the boot issue. I loaded them and power cycled. RC is still working fine.

tops4u, 2019-01-16:
So my Unit is just different then - or I'm handling it wrong :-/

Those are two different Problems. With 1.8.1/1.8.2 and the mentioned Parameters the Unit will just not finish the boot process but RC works and the IO is reachable. 

I have not checked wether I get RC on the newly initialized Device on Master. I'll check this today.

davids5, 2019-01-16:
@tops4u - It is hard to say. As I said I did load your params. I had only USB, JTAG and and the XSR connected. 

To debug it, You and I can get to a common minimal state with `mtd erase` and `param set SYS_AUTOSTART 4011` 

Please ask @dagar to help you find the serial port mappings that had been in the rcS. Then see if you have something using USART6 in you params.

tops4u, 2019-01-16:
@davids5  & @dagar 

Found it! Sorry it was stupid. I was using the same SD I had in the mRo X2.1 before. For the OSD I have put the following line in extras.txt: mavlink start -d /dev/ttyS6 -b 57600 -m osd -r 10000

This caused all the havoc with RC In and also with the communication with the IO. Strange still, that it did not happen with 1.8.2

So a big **sorry** for the inconvenience, and thanks a lot for the help!