PX4/PX4-Autopilot #10002: Windows CI: Appveyor based SITL build using Cygwin Toolchain

MaEtUgR, 2018-07-23:
As requested by @dagar and proven to be necessary multiple times: I want to add CI build support on Windows such that it's easily visible to developers when the build breaks in the Cygwin toolchain described here: https://dev.px4.io/en/setup/dev_env_windows_cygwin.html

References: #9981 , #9920

MaEtUgR, 2018-07-23:
From #9981
> **Update:** I worked on the [CI using Appveyor](https://github.com/MaEtUgR/Firmware/blob/pr-cygwin-ci/appveyor.yml), it only doesn't build master yet because something with numpy on the latest cygwin is buggy [and it doesn't import correctly](https://ci.appveyor.com/project/MaEtUgR/firmware-b5mk6).
And I still have to figure out how to checkout the correct branch. But it looks like I'm almost there thanks to the [script based toolchain setup repo](https://github.com/MaEtUgR/PX4Toolchain) I did before which only required some small adaptions.

dagar, 2018-07-23:
Appveyor is likely the best solution if possible. It also serves as a nice test for the installer.

If that's not workable for whatever reason, then worst case we could add a Windows machine to Jenkins. Someone would need to manually setup (and maintain) the dev environment though.

MaEtUgR, 2018-07-23:
I did it with scripts always installing everything fresh from the internet right now. I can do the installer later as well.
> Someone would need to manually setup (and maintain) the dev environment though.

I could easily do that.

MaEtUgR, 2018-07-26:
@dagar 🎉 finally I have the first successful appveyor build of sitl and px4fmu-v4:
https://ci.appveyor.com/project/MaEtUgR/firmware-b5mk6/build/23
Had to find multiple good hiding script details to make it work. Now I create a pr and explain more there.

closed 2018-07-26 by LorenzMeier