PX4/PX4-Autopilot #10015: low ADC readings rate

dollop80, 2018-07-24:
My module logic is based on adc readings. I used the uORB superblock class to implement the module and subscribed it to adc_report subscription. The first 5 seconds after reboot adc_report returns data with high data rate = 100Hz. In a 5 seconds after startup the adc_report publishes data with the same rate, but the values have about 3-4Hz update frequency.

**To Reproduce**
subscribe to adc_report and publish debug output to see the values.

**Expected behavior**
adc values from adc_report have 100Hz update rate

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=fae5ec99-843c-473f-9f22-83f309ce987c 
[image]

LorenzMeier, 2018-07-24:
Depending on your radio link it might be just the transmission. I would recommend looking at a .ulg log file from onboard.

dollop80, 2018-07-24:
It is not a transmission problem. Pixhawk connected via USB. Moreover I sent the data to servo output (mixed with IMU readings) and the servo react smoothly on IMU input and performs "darts" with 3-4Hz update rate depending on adc readings

LorenzMeier, 2018-07-24:
Then you likely caused a busy cycle and too high CPU load. Run ‚top‘ in the shell in QGC. None of this is a core system issue, so closing this. Feel free to file a new issue of you have code that shows that there is a software fault outside of your own changes.

closed 2018-07-24 by LorenzMeier

dollop80, 2018-07-24:
CPU usage is 40%.
The code is from segway example. Only the subscription to adc_report is added.
[image]