PX4/PX4-Autopilot #10016: Altitude hold with LPE on pixhawk not stable as it should be

kyuhyong, 2018-07-24:
**Describe the bug**
A clear and concise description of the bug.
git cloned latest firmware master and build for LPE support with following setup
```
        #
	# Estimation modules
	#
	modules/attitude_estimator_q
	modules/ekf2
        modules/local_position_estimator
	#modules/position_estimator_inav
	modules/landing_target_estimator
	#modules/wind_estimator
```
Set SYS_MC_EST_GROUP to use LPE
Set LPE_FUSION to 160 to enable "pub agl as lpos down, fuse baro"
Set SENS_EN_LL40LS to PWM to enable sonar data
Sonar pwm input is connected to AUX5 channel and checked sonar data reading.
No problem in Stabilized flying.
Issue is
Altitude hold is not as stable as it should be. Drone ups and down with throttle input at around 50%. 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on in indoor with flat floor.
2. Take off with stabilized mode on.
3. Switch to Altitude mode at about 1m above ground while throttle is about 50%.
4. Drone moves up and down by about 50 cm continuously.

**Expected behavior**
Drone should be stable at certain height within 10 cm up and down. 

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=67c9cc5b-8b89-4b54-b565-db19b0fa8db6

Add screenshots to help explain your problem.
[plot](https://ibb.co/jVC9m8)

**Drone (please complete the following information):**
- Describe the type of drone.
 Quadcopter with 280 mm diagonal
- Photo of the IMU / autopilot setup if possible.

LorenzMeier, 2018-07-24:
Mainline support is only offered for EKF2. @TSC21 might still be interested to help though. I’m closing this as unsupported.

closed 2018-07-24 by LorenzMeier

kyuhyong, 2018-07-24:
Does EKF2 supports px4flow for position hold?

bkueng, 2018-07-24:
Yes. You need to enable it in `EKF2_AID_MASK`