PX4/PX4-Autopilot #10027: Wrong gyro bias can cause bad rate control

AndreasAntener, 2018-07-25:
With the current sensor pipeline the rate control loops have a dependency on the estimator. Before we considered rate control to be "safe" even if there is an estimation problem.

The dependency is the use of the gyro bias coming from the estimator, e.g. in mc_att_control: https://github.com/PX4/Firmware/blob/master/src/modules/mc_att_control/mc_att_control_main.cpp#L529-L531. The same is true for fixed-wing but the bias is added in EKF.

@dagar @priseborough 
- Could we publish something from EKF to invalidate the gyro biases? (do we know when they are not valid?)
- Could we set hard limits to these biases?

We ran into this problem by flying outdoors with EKF configured for indoors and we ended up having gyro biases >0.2rad/s which gave the pilot quite something to do ;)

priseborough, 2018-07-25:
Bias states are limited internally within the EKF in the constrainStates() function to an equivalent 20 deg/sec which is large. We could probably reduce that down to 5 deg/sec on the assumption that the gyro is calibrated at a nominal operating temperature and drift is only due to variation about that point.

AndreasAntener, 2018-07-25:
Sounds good. To be fair it was an issue before already, but the limits were very low too (0.05rad/s).

closed 2019-02-18 by stale[bot]