PX4/PX4-Autopilot #10033: Yaw during takeoff during first flight after reboot

DanielePettenuzzo, 2018-07-25:
Every time I take off for the first time after a reboot I get some strange yaw behaviour. As soon as the vehicle takes off I get a very fast yaw (more than 90 degrees). If I land and do a second flight it doesn't happen anymore (probably some initialization issue).

Frame: S500
Autopilot: Pixhawk 4 Mini (fmu-v5)
PX4 version: Today's Master (100f9553cef2203c818ab98e13d39406c6fc3c8c)

Position Mode:
Log: https://review.px4.io/plot_app?log=fe704b1d-0e77-4bc4-9c9f-35aa2da2732c
Video: https://youtu.be/A_9ktAQalvs 

*Log and video are from the same flight
*Log is 'from boot to disarm'

DanielePettenuzzo, 2018-07-25:
FYI @MaEtUgR @Stifael @RomanBapst

Stifael, 2018-07-25:
@DanielePettenuzzo thank for input.
I think the reason why it happens is that the estimator has increased the reset counter by one before a takeoff.  If that happens, then [this](**https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/FlightTaskManualStabilized.cpp#L99**) is true and the yaw setpoint is added by an offset. 

I think I know how to fix that such that the reset will not cause any issue even if it occurs before takeoff

Stifael, 2018-07-25:
this [PR](https://github.com/PX4/Firmware/pull/10035) should fix it if I am right with the assumption given above. 
Unfortunately, right now I haven't found an easy way to force the estimator to reset. @priseborough do you have a good idea how to do that in simulation?

Stifael, 2018-07-25:
@DanielePettenuzzo if you like and have time, do you mind to test #10035?

DanielePettenuzzo, 2018-07-26:
@Stifael Sure! I can fly later today.

commit fc8a05f6 (2018-07-28): estimataor reset counter: set Flighttaskcounter equal reset-counter during activation (#10035)

 - fixes #10033
closed 2018-07-28 by dagar