PX4/PX4-Autopilot #10034: Pixhawk 4 Mini (fmu-v5) minor issues

DanielePettenuzzo, 2018-07-25:
Autopilot: Pixhawk 4 Mini (fmu-v5)
PX4 Version: Today's Master (100f9553cef2203c818ab98e13d39406c6fc3c8c)

Minor issues while setting up board for flight:

- By default the SYS_USE_IO param was set although the board doesn't have an IO. The firmware doesn't ignore this param therefore I don't have my pwms on the fmu port. When I ran the 'pwm info' command I got a hardfault.

- During boot up the buzzer inside the Pixhawk 4 GPS makes a strange sound instead of the regular sound from the normal pixhawk 4.

- PWR LED (green) is blinking and ACT LED (blue) is off during normal operation. Should the PWR LED always be on and the ACT LED blink? This changed on master in the last couple of weeks. Before PWR was always on and ACT always off.

DanielePettenuzzo, 2018-07-25:
FYI @davids5 @bkueng

jinger26, 2018-07-27:
FYI @jamming

davids5, 2018-08-16:
@DanielePettenuzzo 

I do not have the final rev of this FMU. (@jinger26 any eta?) So I do not have labeling on the LEDs. I can tell you from what my FULL FMUv5 we are indeed blinking the pwr LED (see below). But on the REV of the FMUv5 mini I have, with the PWM connectors pointing up, USB to the right, the LED's are G R B. However the R and G LED are swapped (are on different pins)  to fmuv5. So my red led is blinking.

FYI @jamming, @jinger26 To be clear This is either an assembly issue or lay out bug on the version and as built mini I have.  

```
led_on(LED_RED) is turning on GREEN
led_on(LED_GREEN) is turning on RED
```

Once all the HW is correct we need to fix the logical use of the LEDs.

The Labeling is 

**Label**|**Color**|**Bootloader usage**|**FMU usage**
-------------------------------|-----------|-------------------------|-----------------------------------
ACT - Activity|Blue|Flutters when the bootloader is receiving data|Indication of ARM
B/E - In Bootloader / Error|Red/Amber|Flutters when in the bootloader|Indication of an ERROR
PWR - Power|Green|not used by booloader|indication of ARM state

**Red/Amber**|**Blue**| **Green**|** Meaning **
-----------------|---------|-----------|-----------------
10Hz  | x|x| Overload CPU load > 80% or RAM usage > 98%
OFF |x|x| Overload CPU load <= 80% or RAM usage <= 98%
NA| OFF| 4 Hz| actuator_armed->armed && failsafe 
NA| ON| 4 Hz| actuator_armed->armed && !failsafe
NA| OFF|1 Hz|!actuator_armed->armed && actuator_armed->ready_to_arm
NA| OFF|10 Hz|!actuator_armed->armed  && !actuator_armed->ready_to_arm 

x - is any state

The source of status LED control (and color choice) is in Commander. There is no abstraction, this will need to be remedied. The Boards should provide a set of defines or const array for translation by indication OVERLOAD_LED, ARMED_LED, to the color then the code can "activate" the indication on the OVERLOAD_LED and that can be whatever LED the board chooses. The latter cost more FLASH bit is more appealing to me.. 

As secondary issue, I discussed with @dagar the interface to the LEDs are a RW-IOCTL interface. While this is a proper user/kernel separation and an abstraction. There is a lot code to enact it. 

perhaps we should consider a re-factor

```
#define SET_OVERLOAD_LED(on_off) 
#define TOGGLE_OVERLOAD_LED() 
```
or
```
board_set_status_led(int logical_led_function, int state {on off, toggle}) 
```
The macros could be a direct GPIO manipulation on a platform that supports it, or an IOCTL on one the does not.

The call able could be a translation and then a direct GPIO manipulation on a platform that supports it, or a translation and IOCTL on one the does not.

Thoughts?

jamming, 2018-08-20:
On the final rev of FMUv5 mini and FMUv5 pro, the definition of LED are the same.

Green      --- PC6 --- PWR
Amber   --- PB1 --- B/E
Blue        --- PC7 --- ACT

davids5, 2018-09-26:
@DanielePettenuzzo - see LEDs are fixed here https://github.com/PX4/Firmware/pull/10579
PWR is on solid and ACT indicates the arm state.

davids5, 2018-09-26:
@DanielePettenuzzo the SYS_USE_IO is here https://github.com/PX4/Firmware/pull/10580

davids5, 2018-09-26:
@DanielePettenuzzo / @PX4/testflights - do you have FMUv5mini RC05?

Please cross test that

Leds is fixed
SYS IO set ok.

>During boot up the buzzer inside the Pixhawk 4 GPS makes a strange sound instead of the regular sound from the normal pixhawk 4.

This should have been fixed in master when the tone_alarm/tune control came in.

(rebase the PRs into one to test)

davids5, 2018-09-26:
Bug fix fmu hard fault on pwm info was in b578419e0950994b4e0a002362e294b3fc74fca6

DanielePettenuzzo, 2018-09-26:
@davids5 yes I have the RC05. I can test these.

jinger26, 2018-11-20:
Are we able to close this issue now? @davids5 @dagar

davids5, 2018-11-20:
@jinger26 the PR listed above are merged. So I will close this. What is left is the back port.

closed 2018-11-20 by davids5

jinger26, 2018-11-21:
@hamishwillee do we need to document the LED labelings above?

davids5, 2018-11-21:
In FMUv5 I have formally defined UI LED and Status LEDs as an explanation of the Legacy LEDs that were on FMUv2 and FMUv4 HW. 

The Super Bright RGB LED is the UI LED. (The normal LED) The discrete LEDs (or not so bright RGB LED on FMUv4)  are the status LED's as marked on the HB FMUv5 (pixhawk 4) case these are the status LED for the PX4IO and FMU SoC and indicate Power, bootloader mode and activity and Errors. 

Green      --- PC6 --- PWR
Amber   --- PB1 --- B/E
Blue        --- PC7 --- ACT

See the Status LED and UI LED definition it the pinining:

https://docs.google.com/spreadsheets/d/1-n0__BYDedQrc_2NHqBenG1DNepAgnHpSGglke-QQwY/edit#gid=912976165

davids5, 2018-11-22:
@hamishwillee - miss-typed your name above. Please see https://github.com/PX4/Firmware/issues/10034#issuecomment-440787304