PX4/PX4-Autopilot #10036: Safety switch not changing its state

shabhu18, 2018-07-25:
my system
pixhawk 2.1
px4 version : 1.8
gps+compass+safety switch is connected in GPS1 port of pixhawk 2.1

Issue : The safety switch doesn't changes its state to arm state. No console output .

how to debug this stuff.

dagar, 2018-07-25:
Can you check the circuit breaker parameters?

To debug start with `listener safety` via nsh.

santiago3dr, 2018-07-25:
I ran a test on a 2.1 cube
flashed 1.8.0 stable and then current master
both look to be engaging the safety switch properly
[image]
[image]

@shabhu18 what are the leds on the GPS module doing?

shabhu18, 2018-07-26:
i get the following if i so listener safety

TOPIC: safety instance 0 #1
 safety_s
	timestamp: 361180857 (0.021590 seconds ago)
	safety_switch_available: 1
	safety_off: 0
	override_available: 0
	override_enabled: 0

The state doesn't change even if i change safety switch for a while.

meanwhile on QGC i get the following
[11:08:11 522] Emergency: Mag 0 fail : Timeout
[11.08.25 512] critcal: PREFLIGHT FAIL :MAG SENSOR INCONSTANT

santiago3dr, 2018-07-26:
given mag 0 is giving error,s it might be a problem with the connector or GPS module... does the gps portion work properly?

shabhu18, 2018-08-29:
I disabled the safety switch in software it worked .

closed 2018-08-29 by shabhu18