PX4/PX4-Autopilot #10040: Dive when switching to AltHold with MPC_THR_HOVER set to 0.6

ecmnet, 2018-07-25:
**Describe the bug**
Setting MPC_THR_HOVER to 0.6, my quad hovers at RC channel 0.5 (a bit overloaded vehicle). When I now switch to AltHold (with LIDAR attached), the thrust setpoint is set to 0.5 which causes a drop of the vehicle. After correction by RC (and going back to RC = 0.5), AltHold works perfectly. Position mode is set to 'smooth'. 

**Expected behavior**
I assumed that with MPC_THR_HOVER = 0.6, the set point of thrust is set to 0.6 instead of 0.5.

**Log Files and Screenshots**
I will share log later. 

[image]

dagar, 2018-07-28:
Which version of PX4?

ecmnet, 2018-07-30:
It was latest master 1.8 , FMUv5 (PixHawk 4).

ecmnet, 2018-07-31:
Works with latest master

closed 2018-07-31 by ecmnet