PX4/PX4-Autopilot #10042: Problem with STATUSTEXT MAVLink message

okalachev, 2018-07-25:
[This commit](https://github.com/PX4/Firmware/commit/eb5b153d1039da2af9a05ff923a185274fa7b19d) adds checking for `is_connected` (= are there heartbeats from GCS) in `STATUSTEXT` sending. But there is a problem.

MAVROS now [has an ability](https://github.com/mavlink/mavros/pull/972) to listen for statustext messages through `/mavros/statustext/recv` topic. It does emit some heartbeats, but only with [the type](https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/sys_status.cpp#L836) `MAV_TYPE_ONBOARD_CONTROLLER`. But PX4 [accepts](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1858) only `MAV_TYPE_GCS`. 

So this check make it doesn't work, while `STATUSTEXT` messages may be useful for user applications (for autonomous drones, etc).

I propose either removing `is_connected` checking, or adding `MAV_TYPE_ONBOARD_CONTROLLER` to the accepted heartbeats types (as in todo note in MAVROS code).

@LorenzMeier , @TSC21 , what do you think?

TSC21, 2018-07-25:
@okalachev that's something I already started but didn't get the chance to finish: https://github.com/PX4/Firmware/pull/7985

okalachev, 2018-07-25:
Maybe it's easier to remove this `is_connected` to fix it for now?
Is it necessary?

TSC21, 2018-07-25:
I do think having a check for the onboard link is the way to go. It should be checked as it was a ground station link.

okalachev, 2018-07-25:
So you mean to add `MAV_TYPE_ONBOARD_CONTROLLER` to accepted heartbeats types, or to wait until your PR is finished? :)

LorenzMeier, 2018-07-25:
Add the heartbeat type please. Thanks!

TSC21, 2018-07-25:
> So you mean to add `MAV_TYPE_ONBOARD_CONTROLLER` to accepted heartbeats types, or to wait until your PR is finished? :)

Well basically they are two sides of the same coin. If you want to bump up my PR, please feel free to do it. I am currently swamped with other stuff.

AlexisTM, 2018-11-23:
One temporary solution that has no impact is to send two heartbeats in mavros: `MAV_TYPE_ONBOARD_CONTROLLER` and `MAV_TYPE_GCS`.

As the first one is sent natively by Mavros, the following snippet sends the second one and enables statustext topic (allowing calibration from mavros for example)

https://gist.github.com/AlexisTM/4a825e1a3a3bbc83f8e1ebc295adeb61

okalachev, 2018-11-23:
@AlexisTM, I did the same here (in C++): https://github.com/CopterExpress/clever/blob/master/clever/src/rc.cpp#L78.

closed 2019-03-09 by stale[bot]