PX4/PX4-Autopilot #10045: 1.8: Can't Calibrate Sensors While UAVCAN is Enabled

keenanjohnson, 2018-07-25:
I get an out of memory error if I do sensor calibrations while UAVCAN is enabled.

This did not happen in 1.7

**To Reproduce**
Steps to reproduce the behavior:
1. Turn drone on
2. Enable UAVCAN
3. Attempt to do sensor calibrations

**Drone (please complete the following information):**
- PX4 1.8
- Pixhawk 2.1 Cube

mhkabir, 2018-07-25:
Simply not enough RAM left, I'm afraid. You could try to disable stuff like extra mavlink instances in the startup script to free up some RAM.

dagar, 2018-07-25:
@keenanjohnson if you can share the output of `top` we might have some ideas.

keenanjohnson, 2018-07-26:
Here's the output of top: 

[image]

LorenzMeier, 2018-07-26:
@keenanjohnson I think Daniel also meant the output of `free` in addition. Thanks!

An obvious win would be the uavcan fw srv. It is used to upgrade nodes like ESCs with new Firmware and we could in theory shut it down before starting any calibration.

dagar, 2018-07-27:
@keenanjohnson the easiest way to save some memory is limit the number of mavlink instances (`mavlink status`). For example when you have USB connected it starts an extra mavlink instance. As a side note you probably shouldn't calibrate the compass with USB attached regardless of memory.

1. don't use USB except when you need it for upgrade or debugging
2. Do you need mavlink on both telem ports? Review `SYS_COMPANION`.
3. Shutdown uavcan firmware update server if you're not using it. `uavcan stop fw` (@LorenzMeier's comment).
4. `uavcan shrink`
5. Consider moving to a pixhawk 4 (or any fmu-v5)

mhkabir, 2018-07-27:
Another option is to disable the dynamic node ID and firmware server permanently. Just assign Node IDs to your nodes manually. This is by far the easiest, and what I do on FMUv3 targets these days. See the `UAVCAN_ENABLE` parameter, and make sure you're on the "Manual Config" option.

dusan19, 2018-08-02:
@mhkabir thanks for the instructions, I just wonder how do I assign the node IDs to my nodes? How do I know what are the IDs of my nodes and where do I set them? I can only see one parameter related to this, UAVCAN_NODE_ID, is that what I should change?
Thanks

mhkabir, 2018-08-02:
@dusan19 You set node IDs on the node (e.g GPS, ESC, etc.) itself. Use an e.g Zubax Babel and `uavcan_gui_tool` (https://uavcan.org/GUI_Tool/Overview/).

Side note : UAVCAN-Mavlink parameter bridge seems to be broken currently. Not sure if it's a QGC thing or PX4 thing.

closed 2019-02-18 by stale[bot]