PX4/PX4-Autopilot #10052: RC SBUS

garfieldG, 2018-07-26:
It seems like there are two places that handle rc, One is in px4io driver and one is in px4fmu driver.
It's seems asymmetrical which might cause issues when using boards with no IO cpu, for example in px4io driver the rssi is addressed while in px4fmu it's not (in case of sbus mode) and the parameters that should handle it are compiled only when using px4io driver.

Isn't it more logical change it to one place that handles RC?

LorenzMeier, 2018-07-26:
Yes, we need to consolidate more of that (bringing more into the parsing libraries that we already have for the different protocols). Would you be interested in taking a stab at that?

garfieldG, 2018-08-01:
I can look at it, not sure that I know the code hyrarchy enough in order to do that

closed 2019-02-18 by stale[bot]