PX4/PX4-Autopilot #10053: fetch_file.py is broken

keenanjohnson, 2018-07-26:
It would appear that the [fetch_file.py](https://github.com/PX4/Firmware/blob/master/Tools/fetch_file.py) script is currently broken.

It appears that the script is checking for an "OK" message before receiving the file.

[image]

However, [this commit](https://github.com/PX4/Firmware/commit/0fdd2b9feac05775553a53475b99b9bf6de6c6d3#diff-83bc6fdf121621e05f2288d011d138ddL66) changed that message header from "OK" to  "File size:"

[image]

LorenzMeier, 2018-07-26:
Can you just use MAVLink FTP? If you need that script to work, can you send a PR to fix?

keenanjohnson, 2018-07-26:
What is MAVLink FTP? How do I use that?

LorenzMeier, 2018-07-26:
Widgets -> Onboard Files gives you all files. Otherwise use the log download:

[image]

closed 2018-07-26 by LorenzMeier

mhkabir, 2018-07-26:
You don't need actually need a screen or QGC. You can implement the mavlink FTP protocol in a custom script as needed. https://mavlink.io/en/protocol/ftp.html

As for the broken tooling, it would be great if you could send a PR removing it or fixing it. It seems like an easy fix, either way.