PX4/PX4-Autopilot #10061: EKF2 fusion vision pose + rotate external vision, unstable initialisation

Changliu52, 2018-07-27:
Hi @mhkabir ,
Please see the flight log here with the firmware at 57dfcee
Handholding test. Link here: https://review.px4.io/plot_app?log=7d6c9ab6-e5fc-4e21-86c8-991b27d3518a

EKF2_AID_MASK: vision position fusion + rotate external vision
EKF2_HGT_MODE: vision
EKF2_EV_DELAY: 0.0ms
EKF2_ASP_DELAY: 250ms

1) The local pose took several attempts at the beginning to follow vision correctly, and at the same time it complained about "EKF ev hgt timeout" and switch between vision and baro. It doesn't make sense because vision x, vision y are ok, only vision z has time out?
2) when switching baro, the spike problem in local position z came back. I think there are still some bugs for resetting the EKF at switching.
3) after several attempts, local pose did start to follow vision with an offset transform (which is good), however the local pose tends to wonder around (around 1-2 m), when in the initial several seconds, or when the motion evolve significant yawing. Was this caused by the internal vision offset estimation?
Please let me know your thoughts.

Best Regards,
Chang

**Log Files and Screenshots**
[image]

mhkabir, 2018-07-28:
@priseborough will be able to better help with the EKF internals here.

Changliu52, 2018-07-30:
Thanks. I feel it is possible that the EKF timeout was caused by the delay in the onboard slam. if I increased the ASP_delay to 300ms the initialization takes less attempts before stable. I could be that many subprograms were initialized at the time when slam were initialized thus it has longer delay at the beginning.

I do feel the 1.7.4 is less sensitive to this delay in the beginning. Is there a way to set the buffer longer than maximum 300ms?

Although I am still not sure about this z-axis spikes, these looks like that the z was drifting (it shouldn't even without vision if it switched to baro), and suddenly reset back to vision z.

Thank you in advance.

mhkabir, 2018-08-04:
@priseborough Can you have a look here?

eleboss, 2018-08-24:
@mhkabir 
Hi, 
I have some question about how to use the  external vision aid.

1.Will the rotate external vision done automatically, if not, which parameters need to be set?  
Do I need to publish my camera rotation pose in the /mavros/vision_pose/pose topic to help this rotation?

2.My height is detected by tfmini rangefinder, but the xy position is achieved from slam system. 
However, when the XY pose have a sudden jump, the xy velocity estimation of ekf2 will be heavily influenced by this jump and given a very unreasonable xy velocity estimation, I am using setpoint_raw/local velocity control, and this sudden jump will directly lead to a unstable of my drone then crush, sad...
So are there any ways to relieve this?

From my perspective:
1.Increase the EV noise may help(Haven't test)
2.Make my slam more accurate(Going on, a low pass filter may help) 
3.Ask you for help

Thank you in advance.

eleboss, 2018-08-24:
BTW, Is any official guide of EV aid available?

closed 2019-02-18 by stale[bot]