PX4/PX4-Autopilot #10065: Offboard: support attitude + altitude setpoint

jlecoeur, 2018-07-27:
**Describe problem solved by the proposed feature**
It would be nice to be able provide {attitude + altitude} offboard setpoint. A common use case is to give a constant altitude setpoint and to vary the attitude setpoint.

**Describe your preferred solution**
This could be integrated in the recent effort to add more offboard control options to the SDK (#10031) @Stifael. But should also be supported on the MAVROS side @TSC21.

**Describe possible alternatives**
Attitude + thrust is already supported, so it is possible to control the altitude from the companion system. However, as PX4 is able to control altitude, why not just give altitude setpoint and let it do it.

Stifael, 2018-07-30:
> It would be nice to be able provide {attitude + altitude} offboard setpoint. A common use case is to give a constant altitude setpoint and to vary the attitude setpoint.

That can easily be added. Once #10031 is merged, I will add another set of supported offboard setpoints.

jlecoeur, 2018-07-30:
Great! Let me know if you need someone for testing.

jlecoeur, 2019-02-04:
This feature is waiting on #10031

closed 2019-02-18 by stale[bot]