PX4/PX4-Autopilot #10078: Feature Request: improved minimum ground speed controller for Fixed Wing

Antiheavy, 2018-07-29:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](http://slack.px4.io/).

Currently the FixedwingPositionControl will increase the airspeed set point to attempt to maintain a minimum ground speed derived from FW_AIRSPD_MIN:  https://github.com/PX4/Firmware/blob/50ae8c9c6f8e5d56e91d43604950ab97333a7e17/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L522

This is somewhat arbitrary.  Users and integrator should be able to explicitly set a minimum ground speed as a separate parameter appropriate to their vehicle and use cases.  Being able to explicitely shoose a minimum ground speed  will help with optimizing and trading endurance, range, takeoff and landing performance and other factors.

We could either add a new parameter that:
a) sets this explicitly, e.g.  **FW_GNDSPD_MIN** (default 5 m/s ?)
b) scales minimum ground speed to minimum airspeed, e.g. **FW_GNDSPD_MIN_SC** (default 1.0), this may be an approach to not impact current users.

Additionally, we should consider adding a moving average filter to the inputs to this calculation to smooth out wind gusts or other temporary disturbances.

FYI @dagar

tstastny, 2018-07-30:
@Antiheavy is the intent here to have this for people who don't use airspeed sensors? Otherwise, couldn't one set the airspeed min with respect to the (already quite low-passed) wind speed estimate from ekf2? With only a ground speed minimum without context of orientation with respect to the wind vector, and also as it is currently implemented with the "ground speed undershoot" -- it is quite an inefficient way of keeping a ground speed (e.g. it will increase the airspeed reference even in the case of flying *with the wind, which would not actually be needed).

I've actually had a similar pr sitting around from a few months now (it's been on my laundry list of things to do, just haven't had it in the top priority) which could handle *some of this. See https://arxiv.org/pdf/1804.04209.pdf which explains the logic, and https://github.com/tstastny/Firmware/tree/pr-l1_updates which is the latest version of the pr. I think there are probably still a couple bugs in the code -- I was having issues with HIL until just the last few days (now actually was just simulating a similar guidance logic yesterday in HIL with wind speeds over airspeed, seems to work). But basically what you could do on top of the pr I've already written (but not pushed) -- is make a minimum ground speed in the airspeed compensation function instead of the currently implemented ground speed = 0 set point used for over-wind scenarios. Then this would both efficiently handle orientation with respect to wind, and also keep your desired ground speed min.

Maybe this feature request is motivation now for me to wrap up that PR finally ;) -- let me know what you think.

chamois94, 2018-08-03:
@tstastny I'm very interested in your PR. Is it something we can see in master soon ? Can we help with that ?

Antiheavy, 2018-08-03:
@tstastny I read your paper, very interesting.  This could be a great improvement over the current implementation of L1 which has problems in high winds, especially trying to get on track in a loiter.  I'm not sure this addresses the minimum ground speed issue directly though, or did I miss it (very likely I missed it)?

A minimum ground speed function is important for the following reasons:

1) ability to Return safely, even in high head-winds (mentioned here by @CarlOlsson ).
2) ability to make efficient (time efficient) progress over the ground in mapping applications.
3) ability to make forward progress during takeoff/climbout in high head-winds.
4) ability to make good forward progress during landing approach into high head-winds.

tstastny, 2018-08-04:
@Antiheavy Yes - I understand the importance of maintaining some positive ground speed -- assuming one wants the plane to go anywhere ;)  My point in sharing the paper was twofold:

1) the feasibility function there can be easily extended by adding a minimum desired ground speed to the wind speed, then all calculations can be done the same, and the minimum ground speed is always maintained. I've actually done something similar to this in another paper I'll be submitting soon to a conference.. but I don't have this one "pretty" enough to post online yet ;)

2) the usefulness of the feasibility function is that you only need to increase the airspeed reference when absolutely necessary to maintain your desired ground speed (or in the case of no desired ground speed -- maintaining zero ground speed and waiting for the wind to pass).

Though - I asked about use of airspeed sensors or not, because this will obviously not help in that scenario, where one is not estimating the full wind vector. Then you'd need to default to something even more conservative, also considering the thrust considerations @CarlOlsson mentioned. However -- (personal bias here) I really don't understand why one would not want to have an airspeed sensor on a fixed wing aircraft if you plan to fly in very high winds. And having it in this scenario means that you don't need to consider thrust at all, just increase the airspeed reference and let TECS do what it normally does.

I got a bit distracted with other things due this coming week the past days - but I'll try to throw together a quick screen shot of a simulation on what I mean by maintaining the min ground speed in this manner. Then I guess I need to rebase and recheck the PR branch in HIL again. I'll add the ground speed min functionality to it as well -- only takes like two extra lines of code. At that point, would surely be nice to have some reviews / help testing from interested parties! :) @chamois94

CarlOlsson, 2018-08-06:
Super cool to see some work going into this direction!

> Though - I asked about use of airspeed sensors or not, because this will obviously not help in that scenario, where one is not estimating the full wind vector.

Our experience is that the synthetic sideslip fusion works very well in estimating the wind vector, with the limitation of course that it takes longer to converge and that you need to change flight direction from time to time.

> I really don't understand why one would not want to have an airspeed sensor on a fixed wing aircraft if you plan to fly in very high winds.

Pitot tubes have a tendency to be blocked by water or dirt, so if you relay on an airspeed sensor delivering data you need at least two of them, or a very robust fall back option. (Don't get me wrong, I also think a proper airplane should have a pitot tube, and to have one also measuring AoA and sideslip would be even more awesome;) I just wanted to highlight on the many problems associated with blocked pitots)

> And having it in this scenario means that you don't need to consider thrust at all, just increase the airspeed reference and let TECS do what it normally does

Related to my comment above I don't really understand this. My concern with the current implementation of TECS and the fixed wing position controller is that in order to tune TECS with an overpowered aircraft one needs to reduce the max throttle, leading to the fact that you need to increase the max throttle again if the wind speed is higher than the maximum airspeed that can be achieved with the previous max throttle. 
E.g. lets say that you want the airplane to ascend with 6 m/s in loiter and between way points. Then you set FW_T_CLMB_MAX to 6.0. The throttle needed for that is, say, 73%. So you set FW_THR_MAX to 0.73. Now if the wind speed is higher than the airspeed at 73% throttle the airplane won't make it home.
I think we should decouple the TECS tuning parameters from the (new) parameters defining the trajectory of the mission

tstastny, 2018-08-06:
> I think we should decouple the TECS tuning parameters from the (new) parameters defining the trajectory of the mission

Ahh I see what you are talking about now. Yes - this is true, right now the max airspeed, the max throttle, and the max climb rate are all coupled together. I agree then on that point that decoupling these things should happen. Maybe better to do in a separate PR though? I think all the L1 changes will already be a somewhat large PR -- we could of course reference them to each other. wdyt?

> Our experience is that the synthetic sideslip fusion works very well in estimating the wind vector

If this can somewhat be assumed for any user not using a pitot - this would simplify the PR (not needing to handle too many cases).

Antiheavy, 2018-08-09:
> Pitot tubes have a tendency to be blocked by water or dirt, so if you relay on an airspeed sensor delivering data you need at least two of them, or a very robust fall back option.

@CarlOlsson if you (and others!) are interested in discussing failsafe methods for pitot-static problems feel free to add your thoughts and comments to this related feature request issue:  https://github.com/PX4/Firmware/issues/9869

tubeme, 2018-11-20:
We fly without airspeed and this will give tremendous improvement in our flight.

Antiheavy, 2019-01-04:
related: https://github.com/PX4/Firmware/pull/11142

Antiheavy, 2019-06-25:
Original request has been implemented and heavily tested here: https://github.com/PX4/Firmware/pull/11381#issuecomment-505274188

Antiheavy, 2019-08-07:
This PR https://github.com/PX4/Firmware/pull/11381 has been heavily tested and even released in our custom branch of v1.8.0.  I recommended merging https://github.com/PX4/Firmware/pull/11381

PR #11381 (2019-08-07): fw_pos_control_l1 add new simple min groundspeed
This introduces a new FW parameter `FW_GND_SPD_MIN` to set the desired minimum groundspeed. This replaces existing logic which projects your current airspeed to the ground and has weird coupling with large pitch changes.

Fixes #10078
closed 2019-08-07 by LorenzMeier