PX4/PX4-Autopilot #10081: HIL simulation with Gazebo and Qgroundcontrol doesn't work

AuterionWrikeBot, 2018-07-30:
Hi everybody.
I've succesfully run SIL simulation with Qgroundcontrol-jMAVSim and Qgroundcontrol-Gazebo, controlling the drone in the simulation with a joystick. Now I'm trying to repeat the same simulations but in HIL mode. I've followed this guide " https://dev.px4.io/en/simulation/hitl.html " for the configuration of QGroundControl and simulators (both Gazebo and jMAVSim). I'm using PixHawk flight controller with PX4 Firmware, initally the 1.8 version (latest). Following the guide I've experienced some issues like:
jMAVSim ONLY notices my "Arm/Disarm" commands and even when I move the throttle stick it gives the messages " takeoff detected " or " land detected ". Other commands are not reproduced in the simulation (Drone is still). Searching in forums i found out that 1.8 PX4 version is affected by this bug, and in fact, using the 1.7 version allows me to run HIL Simulation with Qgroundcontrol-jMAVSim without any problems. So NOW I'm using 1.7 PX4 version.
My main problem now is to run HIL with Gazebo, I'm gonna explain what exactly happens everytime (both PX4 1.8 and 1.7 version). I can properly follow every step of the guide (" PX4 configuration " with QgroundControl and " Gazebo Simulator setup ") without any problems but when I connect via USB the PixHawk and then I run Gazebo in HITL mode with the apposite command (wrote in the guide), I see that the GPS led on the flight controller is blue and not green like it is when using jMAVSim in HIL mode. I don't know if this is a part of the problem but is a difference that i noticed. Then when I launch QGroundControl this happens:
If I've done like indicated in the guide: " Under the General tab of the settings menu, uncheck all AutoConnect boxes except for UDP. " , QgroundControl doesn't notice that the PixHawk is connected. And so anything happens.
But if I leave that box checked, QgroundControl connects to the PixHawk and if I use my joystick for "arm/disarm" or to change flight modes, the GPS led of the PixHawk flashes Green and then return Blue (so i think that QGroundControl is able to communicate with the flight controller) but then on the simulator I can't control the drone. I don't know if this is a problem of "communication" between QGroundControl and Gazebo...
Someone can help me?

julianoes, 2019-04-01:
@thomasgubler what's this bot fight about?

thomasgubler, 2019-04-01:
@julianoes the bot is having fun. Seriously: this issue is not even synced to wrike. There is a handful of issues which behave the same. The bot appears to be re-opening them on github but they are not even synced to wrike. I did not yet get to the bottom of why this can happen.

TSC21, 2019-04-01:
Ahah he's fighting back :D

closed 2019-04-01 by julianoes