PX4/PX4-Autopilot #10088: Error running multi_uav_mavros_sitl.launch with gazebo

MatteoTadiello, 2018-07-30:
When I'm trying to roslaunch the multi_uav_mavros_sitl.launch like the devguide [https://dev.px4.io/en/simulation/multi-vehicle-simulation.html]
some process process die. Is every time a different process to die but often is some gazebo or mavros processes

Here there are the log of some execution of the roslaunch
[roslaunch.log](https://github.com/PX4/Firmware/files/2240603/roslaunch.log)

How can I solve this?

Thanks!

lamping7, 2018-07-30:
I can't see what versions of mavros, gazebo, and px4 you're using, but I do see both mavros and gazebo problems. For starters, do you encounter issues with one vehicle i.e. `roslaunch px4 mavros_posix_sitl.launch`

MatteoTadiello, 2018-07-30:
Thanks for the reply.
I'm using this versions:
-gazebo: 9.3.0
-Ubuntu: 18.04
-Ros: melodic
 for the firmware and mavros I'm using the latest versions

If I use only one vehicle I don't have errors  ( only the ODOM error:
 [FATAL] [1532977235.249006505]: ODOM: invalid body frame orientation "frd" )

But the QGroundControl don't find the vechicle connection

lamping7, 2018-07-30:
I don't think I can help much here. Your Ubuntu and ROS versions aren't tested/supported yet and we're generally using gazebo 7 that comes with Kinetic. I suggest hopping off the 18 train and use what's supported: Ubuntu 16.04, ROS Kinetic. On the other hand, feel free to figure it out and post fixes/PRs.

hamishwillee, 2018-07-30:
As @lamping7 says, we're not running any testing on that software stack. YOu should switch to Ubuntu 16.04 and ROS kinetic (which comes with Gazebo 7). 

The instructions you are pointing to should then work. Note however that there is another way to do this which is not yet documented. See https://github.com/acsl-mipt/multiple-sitl/issues/5

MatteoTadiello, 2018-07-31:
Ok thanks, I'll try to use the method used in [acsl-mipt/multiple-sitl#5](https://github.com/acsl-mipt/multiple-sitl/issues/5). Is it already integrated in Firmware code or I have to use the code in the acsl-mipt repository?

hamishwillee, 2018-07-31:
Not merged into firmware.

closed 2019-02-18 by stale[bot]