PX4/PX4-Autopilot #10092: Iridium driver crashes if device not connected

MariliG, 2018-07-30:
How to reproduce:
1. Disconnect iridium device from pixhawk 
2. Set parameter SYS_COMPANION to  "19,200 kbps Iridium"
3. It fails when rebooting 

@acfloria have you experienced this before?

LorenzMeier, 2018-07-30:
This doesn't fix it, but would allow to get out of it: https://github.com/PX4/Firmware/issues/10095

dagar, 2018-07-30:
Any chance you can get the hardfault log? This might be the /dev/iridiumsbd device disappearing (because the driver failed to start) after the mavlink iridium instance has connected.

acfloria, 2018-07-30:
@MariliG can you specify that more and provide me the output of the system console.

Right now I can tell you the following:
- You first have to start the iridium driver before you start the mavlink instance. This needs to be added manually to the extras.txt. It is also important that you add a small sleep between starting the driver and starting the mavlink instance otherwise you get hardfaults when the driver start fails (I know how to fix it and will do it soon). Only changing the `SYS_COMPANION` parameter will not make the full satellite communication subsystem running.
- If there is no modem connected to the serial port when you try starting the driver then the driver start will fail and also starting the mavlink instance will fail, but the bootup process will still complete.
- I am actually still improving the system and will submit some PR's soon to improve the usability.  Also I will submit a documentation about how to use the system to the user guide.

LorenzMeier, 2018-07-30:
@acfloria Would you mind sending a pull request to https://github.com/PX4/px4_user_guide for the current state so people can at least successfully configure it? Or if it is too hard to describe, send the usability fix PR right away?

We know that people in the aerospace community are trying to leverage your work but failed miserably after enabling the parameter.

I found it always very rewarding to see others being able to leverage my work, and I hope you feel the same way about your contributions. With the UX fixes in I'm sure more people can benefit.

MariliG, 2018-07-31:
@acfloria thanks for the reply. 
This issue is was reported by a user, so as soon as we get the output of the system console we will post it here. 

I am looking forward to your contribution :)

acfloria, 2018-08-07:
The PR containing the documentation: [user guide #307](https://github.com/PX4/px4_user_guide/pull/307)

researcher3, 2018-08-17:
Hello everybody,
today we were able to get some output of the system console. Log files are attached.
Now QGC doesn't even recognize the Pixhawk 4 anymore.
I hope these logs help. I can produce more if you want.

[After param reset - Reboot Log.txt](https://github.com/PX4/Firmware/files/2296997/After.param.reset.-.Reboot.Log.txt)
[Before param reset - Reboot Log.txt](https://github.com/PX4/Firmware/files/2296998/Before.param.reset.-.Reboot.Log.txt)
[Boot Log - After param reset.txt](https://github.com/PX4/Firmware/files/2296999/Boot.Log.-.After.param.reset.txt)

acfloria, 2018-08-17:
@researcher3 I cannot see in this log files the attempt to start the iridium driver.

A successful startup looks as follows:
```
INFO  [iridiumsbd] starting
INFO  [mavlink] mode: Iridium, data rate: 5760 B/s on /dev/iridium @ 115200B
```

If it fails:
```
INFO  [iridiumsbd] starting
ERROR [iridiumsbd] modem not responding
WARN  [iridiumsbd] stopped
```

So I guess there must be something wrong your setup. Anyway in the [user guide](https://docs.px4.io/en/advanced_features/satcom_roadblock.html) you can find now instructions on how to set the system up. Also note that you should use the latest PX4 master as I made the startup more robust and fixed starting the driver in using the `SYS_COMPANION` parameter.

researcher3, 2018-08-24:
**Succuess !!**

I followed the user guide so far as to create the extras.txt with the content:

```
# add a sleep here to make sure that the module is powered. Changed from 200000 to 500000.
usleep 500000

#driver. Changed from SERIAL_PORT to TELEM2
if iridiumsbd start -d TELEM2
then
    # start the iridium mavlink instance. Changed from 115200 to 19200.
    mavlink start -d /dev/iridium -m iridium -b 19200

    # Sleep a bit to give Mavlink time to set up
    usleep 100000

else
    echo "WARN: Iridiumsbd driver not started, reboot"
    tune_control play -m "ML<<CP4CP4CP4CP4CP4"
fi
```

The Pixhawk 4 was recognized by QGC again, made the firmware update and booted. Fantastic.

**Reproduction of the failure:**
After full calibration I changed the SYS_COMPANION parameter, since it was still set to OSD (57600 baud, 8N1). I need to change it to Iridium Telemetry (19200 baud), but the drop-down just offers me Iridium Telemetry (115200 baud, 8N1). Going to Advanced settings -> Manual Entry and change it to 19200 will not do the trick (after Pixhawk reboot). By playing around with different values from the drop-down list (a Pixhawk reboot after each change) for 4 or 5 times I chose Iridium Telemetry (115200 baud, 8N1) and changed the manual entry to 419200 (since the 4 seems to indicate Iridium). After this the Pixhawk 4 didn't reboot as the 4 or 5 times before, but got stuck at the same place (after Firmware upgrade). Same problem as before.

**My solution** (not very elegant):
Replacing the extras.txt with the config.txt (param reset param save) on the SD card, booting it, and swap  the txt files again made the trick again and the Pixhawk runs again. Of course, every setting is default and you have to reconfigure everything.

How can I set the SYS_COMPANION to Iridium Telemetry (19200 baud)?

MariliG, 2018-09-17:
@acfloria do you have any input on the question above?

researcher3, 2018-09-18:
The content of the `config.txt` is:
`param reset
param save`
I got the latest master with the built-in waiting time for the driver.
How much effort would it be to support `SYS_COMPANION` with Iridium 19200?

acfloria, 2018-09-18:
Not too much from a programming side but a lot for in the testing and currently I don't have the time for that. But changing the baudrate on the module is quite easy to do.

acfloria, 2018-12-04:
@MariliG @researcher3 any updates from your side. Could you fix it?

acfloria, 2018-12-04:
Ok, so just to confirm you are using now the iridium module and it is operational?

researcher3, 2018-12-04:
Yes, iridium module is working on default baudrate.

acfloria, 2018-12-04:
Ok, then I am closing the issue as the cause for the hardfault was fixed upstream and the module seems to be working.

closed 2018-12-04 by acfloria