PX4/PX4-Autopilot #10095: Add the ability to reset all parameters from QGC even if system is not any more booting properly

LorenzMeier, 2018-07-30:
**Describe problem solved by the proposed feature**
If you set parameters such that the system doesn't boot properly, the only way to recover is to wipe the storage.

**Describe your preferred solution**
We need some way to reset the parameters on boot as the first instruction.

**Describe possible alternatives**
A dual storage for a configuration that you can switch back and forth.

dagar, 2018-07-30:
"Safe Mode"

dagar, 2018-07-30:
Anyone monitoring FRAM usage?

LorenzMeier, 2018-07-30:
I'm casually monitoring it and we're far off the limits still. We might want to tear down the waypoints partition sometime soon.

davids5, 2018-07-31:
@LorenzMeier I may not understand the issue well.  

Why not just you put an sd card in with '/etc/rc.txt` that has `param reset` 

Would something like this work or do you feel more is needed?

At the top of rcS
check param BOOTING 
if it does not exist create param BOOTING = 1
else if set and == 3 do param reset and reboot.
if it is < 3  the increment it BOOTING = BOOTING + 1
...
at the end of rcS
delete param BOOTING

LorenzMeier, 2018-07-31:
Because whenever I ask people to do that they get it wrong and it doesn't work for them. For a controls engineer only a GUI is a reliable way to do things. They are used to graphically clicking blocks in Simulink, not console commands or making sure files are in the exactly right location with the right name and content.

dagar, 2018-07-31:
Maybe we should talk about the specific use case and workflow?

I'm wondering about...
 - an absolute minimal init that gets you mavlink + parameters (QGC)
 - something physical that could initiate a full reset
 - do we try to detect a bootloop and then do something?

BazookaJoe1900, 2018-08-01:
It might be related to what happened to me at case #10099 [Pixhawk get to Brick mode after selecting Quad H4 680]

closed 2019-02-18 by stale[bot]