PX4/PX4-Autopilot #10099: Pixhawk get to Brick mode after selecting Quad H4 680

BazookaJoe1900, 2018-07-31:
**Describe the bug**
Playing with pixhawk 4 I have selected 'Quad H4 680' airframe on QGC.
After that, the Pixhawk does the start-up tune, but not much more.
There is no communication with the PC Via USB, so I can't change parameters and so.
The only way that I found to recover the Pixhawk is by putting rc.txt on the SD card with 'param reset' commnad on it.

**To Reproduce**
select 'Quad H4 680' airframe on QGC.
Tested twice, no other frame has been tested, or other Pixhawk HW.

dagar, 2018-08-01:
Have you successfully used this airframe before?

BazookaJoe1900, 2018-08-02:
@dagar - No, I don't have this frame and ddidn't use it. I just wanted to look at the parameters of that frame.

closed 2019-02-18 by stale[bot]