PX4/PX4-Autopilot #10102: MAV_CMD_NAV_LAND_LOCAL denied although local position available

ecmnet, 2018-07-31:
**Describe the bug**
MAV_CMD_NAV_LAND_LOCAL is denied with message 'Precision landing denied, land manually' although local position available. Using the command MAV_CMD_NAV_LAND without GPOS sends however the correct message 'Landing at current position' but the vehicle keeps in LOITER.

**Expected behavior**
If LPOS is available MAV_CMD_NAV_LAND_LOCAL should land according to the specified parameters.
Alternatively, MAV_CMD_NAV_LAND should land at current position with LPOS only.

mhkabir, 2018-07-31:
Precision land, as it is implementers currently, needs global position.

I have a branch somewhere where it's hacked to work in local frame only. It's mostly about how the controllers don't process local setpoints when in AUTO mode. (Precland is a submode of AUTO)

ecmnet, 2018-07-31:
What was the idea behind this separate mode? Local landing could be driven either by tags (as precision land does) or by other accurate LPOS sources, like vision. 

@dagar I agree to your statement, that the whole GPOS,LPOS thing needs to be revised - including the possibility to send a GlobalHome to EKF2 and control position locally by then.

AlexKlimaj, 2018-08-02:
I think we just saw this same bug with the latest master. RTL and Land were not getting obeyed.

ndepal, 2018-08-02:
> @ndepal how did you want this to work?

@dagar That appears to be a bug. If `VEHICLE_CMD_*` in `vehicle_command.msg` are supposed to match mavlink's `MAV_CMD_*`, does that mean we can't even really have a `VEHICLE_CMD_NAV_PRECLAND`? How does `MAV_CMD_NAV_LAND_LOCAL` work if it doesn't have a matching `VEHICLE_CMD_*`?

dagar, 2018-08-04:
I believe it should be a superset.

ecmnet, 2018-10-14:
Solved

closed 2018-10-14 by ecmnet