PX4/PX4-Autopilot #10104: CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR

DesmondZhong, 2018-07-31:
Hi!

I was testing Intel Aero RTF with PX4 in POSITION mode and for the most of the time there will be some aggressive behaviors. When I check the logs, the FCU will send back an error. The following is the output of MAVROS:

```
[ WARN] [1532985278.000720962]: TM : RTT too high for timesync: 46.27 ms.
[ERROR] [1532985278.143875280]: FCU: CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR
[ERROR] [1532985278.197775320]: FCU: Failsafe enabled: no local position
[ INFO] [1532985278.245378081]: FCU: Failsafe mode enabled
[ WARN] [1532985278.597816504]: TM : RTT too high for timesync: 42.97 ms.
```

I also write a python script to send waypoints missions through MAVROS. The RTF can take off smoothly but cannot reach the correct waypoint. I got the following error:

```
[ INFO] [1532985449.482349651]: WP: mission cleared
[ INFO] [1532985449.541545251]: FCU: Mission finished, landed.
[ INFO] [1532985449.543720535]: WP: mission sended
[ INFO] [1532985449.625598097]: FCU: Executing Mission
[ INFO] [1532985449.680501045]: FCU: Takeoff to 2.0 meters above home.
[ INFO] [1532985449.728870793]: FCU: Takeoff detected
[ WARN] [1532985450.024027304]: TM : RTT too high for timesync: 43.39 ms.
[ WARN] [1532985450.635279594]: TM : RTT too high for timesync: 54.29 ms.
[ WARN] [1532985452.126519588]: TM : RTT too high for timesync: 45.43 ms.
[ WARN] [1532985453.226403318]: TM : RTT too high for timesync: 45.00 ms.
[ WARN] [1532985454.420344494]: TM : RTT too high for timesync: 39.70 ms.
[ WARN] [1532985455.127577130]: TM : RTT too high for timesync: 46.35 ms.
[ERROR] [1532985455.339997338]: FCU: CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR
[ERROR] [1532985455.424013207]: FCU: Failsafe enabled: no global position
[ INFO] [1532985455.439297617]: FCU: Failsafe mode enabled
[ WARN] [1532985455.722783648]: TM : RTT too high for timesync: 41.41 ms.
```

I wonder why this critical navigation failure error occurs. Is this a PX4 issue or a hardware issue with Intel Aero RTF? It seems the sensors are not correctly calibrated. I follow the instructions on QGroundControl to calibrate the senors. But the error still pops out. BTW: I check the GPS information from QGroundControl before each flight. It seems normal and 3D Lock is enabled.

```
BIOS_VERSION =Aero-01.00.13
OS_VERSION = Ubuntu 16.04.4 LTS"
AIRMAP_VERSION = unknown
FPGA_VERSION = 0xc2
AeroFC firmware version = 1.8.0
```

Let me know if you need any further information. Any clue on why this error could happen helps. Thank you in advance!

dagar, 2018-07-31:
Can you share a flight log? https://logs.px4.io/

DesmondZhong, 2018-08-29:
Hi I'm gonna close this discussion since the issue is not about the firmware. Indeed there are some problem with compass calibration. If you encounter this problem, please calibrate your compass again. After a correctly calibrated compass, this issue disappeared.

closed 2018-08-29 by DesmondZhong