PX4/PX4-Autopilot #10106: px4fmu-v5 (Pixhawk 4) Firmware Unresponsive to PX4FLOW Optical Flow Sensor

umgram1, 2018-08-01:
Hi, 
I work with px4 on a daily basis at my job; I have previously gotten the PX4FLOW optical flow sensor to work and show live data and sensor-specific parameters in px4 firmware (1.8)  on the Pixhawk 2.1 and the Pixhawk Mini. However, with the Pixhawk 4 (firmware px4fmu-v5), connected with exact same cable on I2C, I have no extra "flow" parameters or data. When I type "px4flow status" in the mavlink command, it responds "driver not running." I have narrowed it down to a problem specific to the Pixhawk 4 firmware, as all of my other Pixhawks immediately recognize and initiate optical flow from the PX4FLOW. Anyone used PX4FLOW successfully with Pixhawk 4 px4fmu-v5 firmware? 
regards

notYourUsername, 2018-08-01:
Something here. Shows up on my Pixhawk Mini, but not on my Pixhawk 4. I’ve tried change the address of the flow and still doesn’t connect.

HLDoo, 2018-10-24:
Excuse me. Which flow sensor did you use? 
I also want to add flow with my pixhawk4 too. And the release ver. says "support crazyfiles2.0 flow deck"
For safety, I have sent an Email to Bitcraze. Hope to get some useful information.

mhkabir, 2018-10-24:
@DanielePettenuzzo Looks like your multi-bus scanning magic is needed here :)

DanielePettenuzzo, 2018-10-24:
Are you using the I2CA port on the pixhawk 4? The scanning of this port in the px4flow driver was added to master after the 1.8 release.

mhkabir, 2018-10-24:
Okay, then closing as already fixed on master.

closed 2018-10-24 by mhkabir

ABDULBASIT1998, 2019-08-31:
hey,
 I am the newbie in drone stuffs, I uploaded the firmware on the board which is .px4 file  on my pxflow but the problem is in the mission planner. it is not showing the screen even after clicking the focus mode....
also on the flight data section in the mission planner it say "BAD OPTO HEALTH" in red colour...
what does that mean? please help me with calibration and further stuffs I am using this first time.

julianoes, 2019-09-03:
@ABDULBASIT1998 please don't necro-bump but open a new issue for your issue.