PX4/PX4-Autopilot #10108: Recover from sensors disconnection

BazookaJoe1900, 2018-08-01:
[The sensors I refer on this issue are the external compass and the GPS.]
An external sensor disconnection is potential failure that can happen. It can happen due to "mechanical" disconnection of wires, some power failure to the sensor or noise over the communication lines. 
Those sensors are critical for flying autonomously and in failure can cause fly-a-way and loss of control over the drone.

From my testing, disconnect and re-connect the GPS or the compass from the AP usually isn't recoverable. Even that in the electrical level the sensor is connected, the AP doesn't use it.

closed 2018-08-01 by mhkabir

BazookaJoe1900, 2018-08-02:
@mhkabir - thanks for the explanation.
I was thinking about situation where disable the sensor can do harm, for example- lossing the GPS reciever for few seconds. there is no redudency usually, and it might be smarter to try to re-use it. It should be with some event to tell the oporator that there is some malfunction wiht the GPS and he should better come for landing.

mhkabir, 2018-08-02:
GPS should recover if reconnected. If not, please file a fresh issue.

BazookaJoe1900, 2018-08-02:
I will re-check the GPS.
What do you think about external compass? what if we don't want to use the internal comapass