PX4/PX4-Autopilot #10109: MPU9250 and BMP280 I2C support

Mtbarros, 2018-08-01:
As mentioned [in this previous issue](https://github.com/PX4/Firmware/issues/5677), support for MPU9250 and BMP280 through the I2C bus is lacking, at least off-the-shelf. According to @dagar 's [comment](https://github.com/PX4/Firmware/issues/5677#issuecomment-409576706), I2C code is already implemented. I have a Pixhawk 1 (without 1MB limit) and a [RY836AI](http://reyax.com/products/ry836ai/)m which has both MPU9250 and BMP280.

Is there a way to make this I2C integration possible? I can help testing in the bench (haven´t started flight trials yet).

Thank you.

mhkabir, 2018-08-01:
Just curious - what is the use case here?

Mtbarros, 2018-08-01:
I am working with an Albatross UAV from [Applied Aeronautics.](https://www.appliedaeronautics.com/) The GPS they provided for assembly was the one mentioned above, RY836AI. Besides the GPS which is already well integrated it combines both these sensors. It is my understanding that connecting these external sensors would improve measurement accuracy. Is this right? I'm just a beginner in this field, so any help or advice is appreciated. Is the use of these external sensors an improvement over using only the internal ones of Pixhawk? Or should I just use the internal ones?

mhkabir, 2018-08-01:
>  It is my understanding that connecting these external sensors would improve measurement accuracy. Is this right? Is the use of these external sensors an improvement over using only the internal ones of Pixhawk?

No, they would be interfaced over I2C, which is inferior to the SPI connection used by the internal sensors. Also only a single set of sensors is used at a time.

Please use the internal ones as is the default. I'm closing this as the use-case for using an I2C IMU doesn't seem compelling.

closed 2018-08-01 by mhkabir