PX4/PX4-Autopilot #10118: Failure to takeoff in Gazebo HITL for fixed-wing airframe

zanemountcastle, 2018-08-02:
**Describe the bug**
SITL work wonderfully on Gazebo for fixed-wing airframes. However, when running HITL on PX4 v1.8.0 using the HILStar plane, the plane refuses to take off, even with a mission loaded, the plane armed and the plane in "Mission" mode.

I have found the PX4 documentation to be somewhat contradictory on the status of fixed wing support for HITL on Gazebo. It is very possible that I am trying something that just doesn't work.

**To Reproduce**
Steps to reproduce the behavior:
1. Following (this)[https://github.com/PX4/Devguide/blob/master/en/simulation/hitl.md] guide, I cloned the firmware, and built Gazebo using
```
make posix_sitl_default gazebo
```
2. With the first terminal still running, I ran,
```
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
```
3. I changed `serialEnabled` and `hil_mode` parameters to true in `Tools/sitl_gazebo/models/plane/plane.sdf`. I also ensured that the `serialDevice` parameter was correct.
4. Finally I quit the two original terminals, opened a new one and ran,
```
gazebo Tools/sitl_gazebo/worlds/plane.world
```
Here is a screenshot of what happens when I run this: 

[image]

Nevertheless, `gzclient` opens and all seems to be working.

5. I opened QGC, connected to Gazebo automatically over UDP, and loaded a mission.
6. I attempted to start the mission and the plane successfully armed, but did not move. Instead, it sat there until I manually disarmed it. 

**Expected behavior**
I expected the plane to simulate an actual takeoff instead of sitting still.

**Log Files and Screenshots**
When I tried downloading the log file, I received an error.

**Drone (please complete the following information):**
- This HITL simulation is running on a Pixhawk 2.1 with PX4 v1.8.0.

roseyangyu, 2018-11-08:
I have same bug to you.Have you solved?

zanemountcastle, 2018-11-09:
Nope, I was never able to figure it out. 😕

TiagoAvelino, 2018-11-13:
I have the same bug too, any discovery, comments here please

akkiconnect, 2018-11-15:
A quick workaround is to use PX4 version 1.7.0 (https://github.com/PX4/Firmware/releases/tag/v1.7.0)
Was able to perform Gazebo HITL of Quadcopter

S259420, 2018-11-27:
@akkiconnect Hello, how did you donwgraded to PX4 1.7.0? Where you able to takeoff the copter using PX41.7.0?

pranavn248, 2018-11-27:
@S259420 You can just download the 1.7.0 firmware from PX4's github and upload it manually using QGC I think. I was able to take off in SITL mode as well when I used this version as.

S259420, 2018-11-27:
@pranavn248 Thankyou for the firmware information. I can takeoff in SITL, the problem is takeoff in  HITL.

TSC21, 2018-11-27:
https://dev.px4.io/en/simulation/hitl.html#compatible_airframe

matteoscanavino, 2018-11-28:
@TSC21 I know that page... HITL is supported for copter with Gazebo. However, I still have takeoff problem...

S259420, 2018-12-02:
@akkiconnect what kind of sensor failures? I didn't experience problems with sensors...

closed 2019-03-16 by stale[bot]