PX4/PX4-Autopilot #10130: Can one do outdoor waypoint navigation with optical flow enabled?

ambikaverma, 2018-08-03:
It seems that GPS and optical flow are independent as per EKF parameter (EKF_AID_MASK).

If optical flow is enabled (EKF_AID_MASK=1) can one do outdoor waypoint navigation still using GPS coordinates? essentially using GPS for getting to waypoints and using optical flow for hovering/ holding position.

Please let me know if my understanding is wrong.
Thanks!

ChristophTobler, 2018-08-03:
>It seems that GPS and optical flow are independent as per EKF parameter (EKF_AID_MASK).

Correct

>If optical flow is enabled (EKF_AID_MASK=1) can one do outdoor waypoint navigation still using GPS coordinates? essentially using GPS for getting to waypoints and using optical flow for hovering/ holding position.

With `EKF_AID_MASK=1`, optical flow is not enabled. Optical flow is enabled if the second bit is enabled (e.g. 2, 3, ...). You can't do waypoint navigation if you only rely on optical flow (2). But you can still do waypoint navigation when using GPS + optical flow (3). So as long as you have GPS you can do waypoint navigation.

ChristophTobler, 2018-08-13:
Please reopen if necessary

closed 2018-08-13 by ChristophTobler