PX4/PX4-Autopilot #10131: IO does not start on master (Pixhawk 4)

RomanBapst, 2018-08-03:
px4io does not seem to be started on latest master. I can manually start it from the shell after boot.
I set the parameter SYS_USE_IO to 1.
There has been quite some activity in the startup scripts recently...

Boot output:
```
[hardfault_log] Fault Log is Armed
INFO  [tune_control] Publishing standard tune 1
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 350216 bytes
INFO  [load_mon] stack check enabled
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
nsh: rgbled_pwm: command not found
MS5611_SPI on SPI bus 4 at 0 (20000 KHz)
WARN  [bst] no devices found
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 0 (1000 KHz)
INFO  [mpu6000] accel cutoff set to 30.00 Hz
INFO  [mpu6000] gyro cutoff set to 80.00 Hz
BMI055_ACCEL on SPI bus 1 at 3 (10000 KHz)
BMI055_GYRO on SPI bus 1 at 2 (10000 KHz)
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [hmc5883] no device on bus 2 (type: 2)
WARN  [hmc5883] no device on bus 4 (type: 2)
INFO  [ist8310] no device on bus 1
INFO  [ist8310] no device on bus 2
IST8310 on I2C bus 3 at 0x0e (bus: 100 KHz, max: 400 KHz)
Airspeed on I2C bus 4 at 0x28 (bus: 100 KHz, max: 100 KHz)
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [mavlink] mode: Normal, data rate: 2000 B/s on /dev/ttyS3 @ 57600B
WARN  [mavlink] hardware flow control not supported
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [mavlink] mode: OSD, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
+ MAV_TYPE: curr: 1 -> new: 21
ERROR [mixer] can't open /dev/pwm_output0

ERROR [mixer] failed to load mixer
ERROR  [init] Failed loading mixer: /etc/mixers/firefly6.main.mix 
INFO  [tone_alarm] missing command, try 'start', status, 'stop'
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
ERROR [pwm] can't open /dev/pwm_output1
INFO  [logger] logger started (mode=all)
px4flow [380:100]
INFO  [px4flow] scanning I2C buses for device..

NuttShell (NSH)
nsh> param INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
show MAV_TYPE
Symbols: x = used, + = saved, * = unsaved
x + MAV_TYPE [347,511] : 21

 1137 parameters total, 712 used.
nsh> param show SYS_USE*
Symbols: x = used, + = saved, * = unsaved
x + SYS_USE_IO [675,910] : 1

 1137 parameters total, 712 used.
nsh> 

```

RomanBapst, 2018-08-03:
@bkueng @mcsauder FYI

RomanBapst, 2018-08-03:
Fixed  by #10132

dagar, 2018-08-03:
Please try master as of https://github.com/PX4/Firmware/commit/a6f93c30d9151d0d913868e74e48979cd89197c8.

mcsauder, 2018-08-03:
Apologies that case got through @RomanBapst , thanks for reporting that.

dagar, 2018-08-04:
Fixed by https://github.com/PX4/Firmware/commit/a6f93c30d9151d0d913868e74e48979cd89197c8

closed 2018-08-04 by dagar