PX4/PX4-Autopilot #10134: no rc detected on pixhawk v4 pro, pixhawk 1 , pixhawk mini v3 and pixhawk 2

dannyfpv, 2018-08-03:
install the new master on the four flight controllers and fail, if use the ac8f4426 version work propely
**To Reproduce**
flash the new firmware and the flight mode go to manual mode.

Add screenshots to help explain your problem.
[image]

[image]

one (please complete the following information):**
the four frames f450 - Photo of the IMU / autopilot setup if possible.

@Tony3dr

dagar, 2018-08-03:
Likely a duplicate of https://github.com/PX4/Firmware/issues/10131.
See https://github.com/PX4/Firmware/pull/10132

dagar, 2018-08-03:
Please try master as of https://github.com/PX4/Firmware/commit/a6f93c30d9151d0d913868e74e48979cd89197c8.

dagar, 2018-08-07:
Fixed in master.

closed 2018-08-07 by dagar