PX4/PX4-Autopilot #10136: exiftool missing from Linux toolchain

DonLakeFlyer, 2018-08-03:
I installed the SITL gazebo toolchain as specified in thge dev guide. When I ran ```make posix gazebo_typhoon_h480``` it told me the exiftool package was missing.

hamishwillee, 2018-08-05:
That's a new one on me. And isn't exiftool also dependent on Perl. @dagar Do you install this in the containers. Don, what platform are you testing this on - Ubuntu 16.04?

dagar, 2018-08-05:
@hamishwillee it can be added to the simulation requirements (both  dev guide and container).

DonLakeFlyer, 2018-08-05:
Ubuntu 16.04. a ```sudo apt-get install exiftool``` fixed it for me.

DonLakeFlyer, 2018-08-05:
Should be included in the linux sim toolchain script.

hamishwillee, 2018-08-05:
As per @DonLakeFlyer info above I added exiftool (`sudo apt-get install exiftool`) into the docs and doc script here: https://github.com/PX4/Devguide/commit/bbbb7a61eee354da326621a06bd9550de12daa33 , https://github.com/PX4/Devguide/commit/b87d0bcf77bff89d1b5cb2172d68122cf6f53244

But @dagar I see you added `libimage-exiftool-perl` as per here: https://github.com/PX4/containers/commit/1e4b966b1ab4d4270a33561da2b396b992530502
What's the different/which is best?

DonLakeFlyer, 2018-08-06:
MY thinking was that it needed to be added here: 
https://raw.githubusercontent.com/PX4/Devguide/master/build_scripts/ubuntu_sim_nuttx.sh

hamishwillee, 2018-08-06:
@DonLakeFlyer The common dependencies that I updated is a base file get downloaded into all the other files. Now I could have put it in that file but chose to put in the base. 

More concerned about the differences between exiftool and libimage-exiftool-perl - we're moving towards a state where the docker container scripts will be used by "end users" instead of the build_scripts so I want them to be as close as possible.

closed 2019-02-18 by stale[bot]