PX4/PX4-Autopilot #10137: static error in altitude setpoint

samirrite, 2018-08-03:
Hi there,
I meet some problem with tecs algorithm, I'm trying to tune it, throttle and pitch looks to work properly in most of cases except one :
When altitude setpoint is a ramp (climbing or descending), i've a static error around 10 meters. Is that due to the first order IIR filter for integrator ? Does it need a second order ? Or maybe there is a couple of parameters that can fix it.
Thank you

chamois94, 2018-08-03:
It seems same altitude tracking error than 3 years ago here https://github.com/PX4/Firmware/issues/2980.

Do you think tuning FF gain will do the trick ?

chamois94, 2018-08-04:
Ardupilot uses an approach removing lag https://github.com/ArduPilot/ardupilot/commit/597273cfffc1d6ae935c6fb40a35cb58626456ed#diff-b6ecb54e20c4b89219bde19a4b8e93ee. I think it is not very elegant. What do you think?

priseborough, 2018-08-05:
Not very elegant, but it works. A more accurate alternative alternative during flare and hold off, is to simultaneously track vertical accln, velocity and position in the height controller, where these are obtained from a PD controller tracking a reference trajectory. The height controller then generates a vertical acceleration demand , which is converted to a flight path rate using TAS. Flight path rate demand is then be converted to pitch rate demand using a lead/lag filter to compensate for the time constant from pitch rate to flight path rate.

chamois94, 2018-08-06:
@priseborough  if I understand correctly you'll change the height rate setpoint into a height acceleration setpoint  by completely modifying  _update_height_setpoint to increase by an order of magnitude 1 the control loop ? Only for landing or in every case ?

chamois94, 2018-08-16:
@priseborough do you think implementing your trajectory tracking for landing slope phase is also a good idea ?

closed 2019-02-18 by stale[bot]